BertaBescos/DynaSLAM #1: make error

csufangyu, 2018-10-27:
thanks for your work! I try to run ,but I meet a problems,could you help me???
     bool mIsKeyFrame;
          ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:127:13: warning:   ‘cv::Mat ORB_SLAM2::Frame::mImDepth’ [-Wreorder]
     cv::Mat mImDepth;
             ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:28:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const Frame &frame)
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h: In constructor ‘ORB_SLAM2::Frame::Frame(const cv::Mat&, const cv::Mat&, const cv::Mat&, const double&, ORB_SLAM2::ORBextractor*, ORB_SLAM2::ORBVocabulary*, cv::Mat&, cv::Mat&, const float&, const float&)’:
/home/robot/slam_project/DynaSLAM/include/Frame.h:129:13: warning: ‘ORB_SLAM2::Frame::mImMask’ will be initialized after [-Wreorder]
     cv::Mat mImMask;
             ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:95:20: warning:   ‘ORB_SLAM2::ORBVocabulary* ORB_SLAM2::Frame::mpORBvocabulary’ [-Wreorder]
     ORBVocabulary* mpORBvocabulary;
                    ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:154:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const double &timeStamp,  ORBextractor* extractor, ORBVocabulary* voc,
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h:130:13: warning: ‘ORB_SLAM2::Frame::mImGray’ will be initialized after [-Wreorder]
     cv::Mat mImGray;
             ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:101:12: warning:   ‘double ORB_SLAM2::Frame::mTimeStamp’ [-Wreorder]
     double mTimeStamp;
            ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:154:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const double &timeStamp,  ORBextractor* extractor, ORBVocabulary* voc,
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h:128:10: warning: ‘ORB_SLAM2::Frame::mIsKeyFrame’ will be initialized after [-Wreorder]
     bool mIsKeyFrame;
          ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:127:13: warning:   ‘cv::Mat ORB_SLAM2::Frame::mImDepth’ [-Wreorder]
     cv::Mat mImDepth;
             ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:154:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const double &timeStamp,  ORBextractor* extractor, ORBVocabulary* voc,
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h: In constructor ‘ORB_SLAM2::Frame::Frame(const cv::Mat&, const cv::Mat&, const cv::Mat&, const cv::Mat&, const double&, ORB_SLAM2::ORBextractor*, ORB_SLAM2::ORBVocabulary*, cv::Mat&, cv::Mat&, const float&, const float&)’:
/home/robot/slam_project/DynaSLAM/include/Frame.h:129:13: warning: ‘ORB_SLAM2::Frame::mImMask’ will be initialized after [-Wreorder]
     cv::Mat mImMask;
             ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:95:20: warning:   ‘ORB_SLAM2::ORBVocabulary* ORB_SLAM2::Frame::mpORBvocabulary’ [-Wreorder]
     ORBVocabulary* mpORBvocabulary;
                    ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:236:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const cv::Mat &imRGB,
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h:130:13: warning: ‘ORB_SLAM2::Frame::mImGray’ will be initialized after [-Wreorder]
     cv::Mat mImGray;
             ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:101:12: warning:   ‘double ORB_SLAM2::Frame::mTimeStamp’ [-Wreorder]
     double mTimeStamp;
            ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:236:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const cv::Mat &imRGB,
 ^
In file included from /home/robot/slam_project/DynaSLAM/src/Frame.cc:10:0:
/home/robot/slam_project/DynaSLAM/include/Frame.h:128:10: warning: ‘ORB_SLAM2::Frame::mIsKeyFrame’ will be initialized after [-Wreorder]
     bool mIsKeyFrame;
          ^
/home/robot/slam_project/DynaSLAM/include/Frame.h:127:13: warning:   ‘cv::Mat ORB_SLAM2::Frame::mImDepth’ [-Wreorder]
     cv::Mat mImDepth;
             ^
/home/robot/slam_project/DynaSLAM/src/Frame.cc:236:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const cv::Mat &imRGB,
[... 247 lines cut ...]
make[2]: *** [../Examples/Monocular/mono_tum] 错误 1
make[1]: *** [CMakeFiles/mono_tum.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o：在函数‘_GLOBAL__sub_I_main’中：
mono_kitti.cc:(.text.startup+0x1473)：对‘boost::system::generic_category()’未定义的引用
mono_kitti.cc:(.text.startup+0x147f)：对‘boost::system::generic_category()’未定义的引用
mono_kitti.cc:(.text.startup+0x148b)：对‘boost::system::system_category()’未定义的引用
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/Monocular/mono_kitti] 错误 1
make[1]: *** [CMakeFiles/mono_kitti.dir/all] 错误 2
CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o：在函数‘_GLOBAL__sub_I_main’中：
rgbd_tum.cc:(.text.startup+0x1f33)：对‘boost::system::generic_category()’未定义的引用
rgbd_tum.cc:(.text.startup+0x1f3f)：对‘boost::system::generic_category()’未定义的引用
rgbd_tum.cc:(.text.startup+0x1f4b)：对‘boost::system::system_category()’未定义的引用
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/RGB-D/rgbd_tum] 错误 1
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] 错误 2
CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o：在函数‘_GLOBAL__sub_I_main’中：
stereo_kitti.cc:(.text.startup+0x1953)：对‘boost::system::generic_category()’未定义的引用
stereo_kitti.cc:(.text.startup+0x195f)：对‘boost::system::generic_category()’未定义的引用
stereo_kitti.cc:(.text.startup+0x196b)：对‘boost::system::system_category()’未定义的引用
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/Stereo/stereo_kitti] 错误 1
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] 错误 2
make: *** [all] 错误 2

csufangyu, 2018-10-28:
I think the reason maybe the version of python! could you help me??

BertaBescos, 2018-10-28:
Hi, it is possible that you do not have installed some of the Mask R-CNN required packages. Run the provided python script `src/python/Check.py` to check if everything needed for Mask R-CNN is properly installed.

csufangyu, 2018-10-29:
Thanks for your answer, and I try to do it.I run the Check.py  and no error. but when I run the code ,the error also happened.And I debug it ,and I meet the error is:
ImportError: numpy.core.multiarray failed to import
And if I run the MaskRCNN.py, it is no such error!could you tell me the version of numpy??I guess it maybe influenced by the version of numpy! thanks very much!

BertaBescos, 2018-10-29:
I am using Python 2.7 and numpy 1.15.2.

csufangyu, 2018-10-29:
thanks very much.But I find the some code is python3，and It does not matter?-------- 原始邮件 --------主题：Re: [BertaBescos/DynaSLAM] make error (#1)发件人：BertaBescos 收件人：BertaBescos/DynaSLAM 抄送：universe ,Author I am using Python 2.7 and numpy 1.15.2.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-433879822",
"url": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-433879822",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [BertaBescos/DynaSLAM] make error (#1)",
"sections": [
{
"text": "",
"activityTitle": "**BertaBescos**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@BertaBescos",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"BertaBescos/DynaSLAM\",\n\"issueId\": 1,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"BertaBescos/DynaSLAM\",\n\"issueId\": 1\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-433879822"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 401940849\n}"
}
],
"themeColor": "26292E"
}
]

BertaBescos, 2018-10-30:
Did you get it to work?

csufangyu, 2018-10-30:
No,I am trying to run，but IT failed!and the nvidia get some error.now I am try to install nvidia!thanks very much.I find some codes are python2，others are python3.-------- 原始邮件 --------主题：Re: [BertaBescos/DynaSLAM] make error (#1)发件人：BertaBescos 收件人：BertaBescos/DynaSLAM 抄送：universe ,Author Did you get it to work?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-434287454",
"url": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-434287454",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [BertaBescos/DynaSLAM] make error (#1)",
"sections": [
{
"text": "",
"activityTitle": "**BertaBescos**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@BertaBescos",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"BertaBescos/DynaSLAM\",\n\"issueId\": 1,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"BertaBescos/DynaSLAM\",\n\"issueId\": 1\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/BertaBescos/DynaSLAM/issues/1#issuecomment-434287454"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 401940849\n}"
}
],
"themeColor": "26292E"
}
]

BertaBescos, 2018-11-05:
I have removed those files that are python3, since they are unused. Thanks for pointing that out.

closed 2018-11-06 by BertaBescos