BertaBescos/DynaSLAM #2: I have same problem

woshilh, 2018-11-02:
Importing Mask R-CNN Settings...
ImportError: numpy.core.multiarray failed to import
Segmentation fault (core dumped)

csufangyu, 2018-11-06:
Maybe you use the anacoda, do not use it! I solve it by use pip.

closed 2018-11-16 by BertaBescos

Chrislzy1993, 2019-03-19:
@csufangyu , 兄弟你好, 我是使用anaconda新建的Python27环境, 然后我根据check.py中的要求安装了依赖库, 同时python check.py没问题, 但是运行出现这个问题. 你说不要使用anaconda是指?????

csufangyu, 2019-03-19:
yes-------- 原始邮件 --------主题：Re: [BertaBescos/DynaSLAM] I have same problem (#2)发件人：Chris_lzy 收件人：BertaBescos/DynaSLAM 抄送：universe ,Mention @csufangyu , 兄弟你好, 我是使用anaconda新建的Python27环境, 然后我根据check.py中的要求安装了依赖库, 同时python check.py没问题, 但是运行出现这个问题. 你说不要使用anaconda是指?????

zhengfxxxxx, 2020-04-13:
> Maybe you use the anacoda, do not use it! I solve it by use pip.

hi, thanks for the information. however i'm afraid i can't abandon anaconda. so i'm wondering if you can tell me more information about the error's cause so i can try solve it. thanks.

zhengfxxxxx, 2020-04-13:
> @csufangyu , 兄弟你好, 我是使用anaconda新建的Python27环境, 然后我根据check.py中的要求安装了依赖库, 同时python check.py没问题, 但是运行出现这个问题. 你说不要使用anaconda是指?????

so did you solved it without abandoning anaconda in the end?  
老哥，所以你后来怎么解决的？

KaninchenM, 2020-07-08:
> > Maybe you use the anacoda, do not use it! I solve it by use pip.
> 
> hi, thanks for the information. however i'm afraid i can't abandon anaconda. so i'm wondering if you can tell me more information about the error's cause so i can try solving it. thanks.

I've used anaconda. Sometimes I'll meet the error `Segmentation fault (core dumped)`, not everytime, not every data sequence. I still can't find and solve the problem. My conda environment can run well most of the time. Did you complete your environment? Maybe I can share my conda/pip requirements with you.