BertaBescos/DynaSLAM #3: make build.sh error

Tacode, 2018-11-05:
Thanks for your work! I try to run the build.sh ,but I meet a problems,could you help me?

error: cannot declare variable ‘DynaSLAM::g_numpyAllocator’ to be of abstract type ‘DynaSLAM::NumpyAllocator’  NumpyAllocator g_numpyAllocator;

Tacode, 2018-11-05:
This problem was solved when the OpenCv2 was installed.

BertaBescos, 2018-11-05:
Ok great! I close the issue.

closed 2018-11-05 by BertaBescos

cumtchenchang, 2018-12-13:
@BertaBescos  I also met this error, but all my code depend on opencv3.  I tried use this code with opencv2.4, the error is still on. Please help me, thank you.

cumtchenchang, 2018-12-13:
error information:
/home/cc/catkin_ws/src/DynaSLAM/src/Conversion.cc:205:16: error: cannot declare variable ‘DynaSLAM::g_numpyAllocator’ to be of abstract type ‘DynaSLAM::NumpyAllocator’
 NumpyAllocator g_numpyAllocator;
/home/cc/catkin_ws/src/DynaSLAM/src/Conversion.cc:74:7: note:   because the following virtual functions are pure within ‘DynaSLAM::NumpyAllocator’:

BertaBescos, 2018-12-13:
Have you checked if it is finding the correct version of OpenCV? These lines in the CMake should tell you what version of OpenCV it is finding:
``message("PROJECT_SOURCE_DIR: " ${OpenCV_DIR})``
``find_package(OpenCV 2.4.11 QUIET)``
``if(NOT OpenCV_FOUND)``
``    message("OpenCV > 2.4.11 not found.")``
    ``find_package(OpenCV 3.0 QUIET)``
  ``  if(NOT OpenCV_FOUND)``
   ``    message(FATAL_ERROR "OpenCV > 3.0 not found.")``
  ``  endif()``
``endif()``

cumtchenchang, 2018-12-13:
I checked, find opencv 3.0. cmake .. sucessfully. However, make with this error

BertaBescos, 2018-12-13:
So the problem is that it is not finding OpenCV 2.4. It is only finding OpenCV 3.0, which is not yet supported.

cumtchenchang, 2018-12-13:
Thank you for your response. May you make your code support opencv3?  OpenCV3 is used often, now.

BertaBescos, 2018-12-13:
Yes, we plan to support OpenCV 3.0 but I am not sure how long it can take us. Many functions changed from OpenCV 2.4 to OpenCV 3.0 in the MatAllocator class. It is in the communication with python where there are a few issues in using OpenCV 3.0 for the moment.

undefinedzero, 2019-02-21:
I found a way to solve this problem. In OpenCV 3.X, **MatAllocator class** has three pure virtual functions and they are const member functions, but in the file src/Conversion.cc, the functions missed the word **const**.

jmfacil, 2019-02-21:
> @BertaBescos I found a way to solve this problem. In OpenCV 3.X, **MatAllocator class** has three pure virtual functions and they are const member functions, but in the file src/Conversion.cc, the functions missed the word **const**.

Thanks! we will add OpenCV 3.X support asap.

Rushm1an, 2021-03-25:
> @BertaBescos I found a way to solve this problem. In OpenCV 3.X, **MatAllocator class** has three pure virtual functions and they are const member functions, but in the file src/Conversion.cc, the functions missed the word **const**.

@undefinedzero how did you solve it? Could you explain a bit?