BertaBescos/DynaSLAM #6: Some questions about running the code

cjh22, 2018-12-04:
Hello! I have some trouble in running the code. Could you please give me some guidance?
1.When I add the argument PATH_TO_MASKS, it always shows segmentation fault. I don't know how to add the argument correctly and what dose the argument mean?
2.When I don't add the argument, it can run but it always can not initialize. I have tried several sequences and it always shows "Trying to Initialize" .
3.I find the MaskSettingsStereo.yaml and should I modify any parameter in the file?

BertaBescos, 2018-12-05:
Hi! Are you having this problem only while running the stereo version of DynaSLAM?

BertaBescos, 2018-12-05:
Hi again!
I realized there was a bug in the stereo example script. It should be working fine now. Thanks a lot for letting me know.

BertaBescos, 2018-12-05:
If you don't add this PATH_TO_MASKS argument, you would be running ORB-SLAM stereo. 
If you add this argument the dynamic objects are detected with Mask R-CNN and the corresponding masks are saved in the provided folder, unless this argument is called no_save. In such case, the masks are used but not written to your disk. I have deleted the file MaskSettinggStereo.yaml since it is unused.

BertaBescos, 2018-12-06:
I am glad it worked! If you run this command `python src/python/Check.py` (with python 2.7) you will have a more detailed output of what packages you might additionally need.

cjh22, 2018-12-06:
OK, I get it! 
I will try to fix it. Thank you again!