BertaBescos/DynaSLAM #10: I have some problem when i runing the code

shejing, 2019-01-24:
when i runing the code with Mask_RCNN, some error occur as follow:
```
nImages: 1352
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
Segmentation fault (core dumped)
```
and i have check Mask_RCNN using Check.py:
```
Initialated Mask RCNN network...
Mask R-CNN is correctly working
```
when i running without MASK_RCNN:
```
Start processing sequence ...
Images in the sequence: 1352

Light Tracking not working because Tracking is not initialized...
Geometry not working.
New map created with 855 points
```
I don't konw why, could you tell me what should i  do ? thank you!

BertaBescos, 2019-01-24:
Hi! Sorry that you are having problems. Can you tell me what example you are running (e.g., rgbd_tum, mono_kitti, etc)?

shejing, 2019-01-24:
it's rgbd_tum

shejing, 2019-01-24:
> Hi! Sorry that you are having problems. Can you tell me what example you are running (e.g., rgbd_tum, mono_kitti, etc)?

it's rgbd_tum

BertaBescos, 2019-01-24:
Mmmm so apparently everything works except the connection between Mask R-CNN and C++... Maybe you are using an OpenCV version different than OpenCV 2.4??

shejing, 2019-01-28:
>           Mmmm so apparently everything works except the connection between Mask R-CNN and C++... Maybe you are using an OpenCV version different than OpenCV 2.4??

My OpenCV version is 2.4.9.....

BertaBescos, 2019-01-30:
I am using OpenCV 2.4.11... That is maybe the reason.

BertaBescos, 2019-01-30:
>Light Tracking not working because Tracking is not initialized...
>Geometry not working.

This does not mean that the software has crashed... This happens only during the first few frames until it succeeds in initializing. 

Still, I do not know why it is crashing when using Mask R-CNN. If Mask-RCNN is working, and DynaSLAM without Mask-RCNN is working, the problem is in the connection between both. Another reason for this crash might the boost libraries: ```sudo apt-get install libboost-all-dev```. Are they correctly installed?

Eman7C7, 2019-01-30:
Yes, Boost libraries are correctly installed. The problem is this one:

> RuntimeError: _ARRAY_API is not PyCapsule object

As you said, it is likely in the connection between the Python part and the C++ one. Unfortunately I am not an expert of Python, nor of its use combined with C++. Therefore I cannot track down the error (and Google seems not to be helpful here).

BertaBescos, 2019-02-04:
I am afraid I cannot really help, since I never ran into this error... 
I read on some related issues that this might happen with python3. In the CMake there are some lines intended to exclusively find the python2.7 directories/libraries:
>include_directories(...
>/usr/include/python2.7/
>/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/
>...)

>target_link_libraries(...
>/usr/lib/x86_64-linux-gnu/libpython2.7.so
>...)

Are these being correctly found?

Eman7C7, 2019-02-04:
Thanks!
The problem was that I was using python3.5. If I force 2.7 everything works!

BertaBescos, 2019-02-04:
That's awesome! I close the issue.

closed 2019-02-04 by BertaBescos

Chrislzy1993, 2019-03-18:
@shejing , did you fix the problem, when u change opencv version???

liubamboo, 2021-04-07:
@Eman7C7 @BertaBescos 
hello, I have the same bug when run the rgbd_tum
""
Reseting Database... done
Light Tracking not working because Tracking is not initialized...
Geometry not working.
New map created with 630 points
""
I try to change the opencv version to 2.4.11 but it doesn't work. And you said force use python 2.7, how can I force to use python 2.7, when I run python command in the terminal, it show python 3.8. when I run python2 command in the terminal,  it show python 2.7. Can you tell the detail how to solve it? Thanks a lot.