BertaBescos/DynaSLAM #11: Something run when I run the code

Gatsby23, 2019-02-18:
I'm sorry to bother you about the Compiling, but here is my problem.
The Check.py shows that it's good

```
/usr/local/lib/python2.7/dist-packages/pycocotools/coco.py:49: UserWarning: 
This call to matplotlib.use() has no effect because the backend has already
been chosen; matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
or matplotlib.backends is imported for the first time.

The backend was *originally* set to 'TkAgg' by the following code:
  File "src/python/Check.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 71, in <module>
    from matplotlib.backends import pylab_setup
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/__init__.py", line 16, in <module>
    line for line in traceback.format_stack()

  import matplotlib; matplotlib.use('Agg')
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Using TensorFlow backend.
Initializing Mask RCNN network...
./src/python
./src/python/logs
./src/python/mask_rcnn_coco.h5
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/keras/backend/tensorflow_backend.py:1154: calling reduce_max (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/keras/backend/tensorflow_backend.py:1188: calling reduce_sum (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
2019-02-18 21:42:14.370602: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
Initialated Mask RCNN network...
Mask R-CNN is correctly working
```

and I correct the CMakeLists.txt to use the opencv2.4.11. It's wired it always shows that I haven't give the mode:
```
 ./mono_tum ~/DynaSLAM/Vocabulary/ORBvoc.txt TUM1.yaml ~/Dataset/rgbd_dataset_freiburg2_desk_with_person /home/robotics/DynaSLAM/src/python/mask_rcnn_coco.h5 
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
mono_tum: /home/robotics/DynaSLAM/src/MaskNet.cc:39: DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->py_module != NULL' failed.
Aborted (core dumped)
```
I also use the python2.7
Could please help me, It makes me crazy.

Gatsby23, 2019-02-18:
the python --version in the terminal is python2.7.12, maybe the CMakelist.txt uses python3.5? But How can I fix it ?
[image]

jmfacil, 2019-02-18:
Hi @Gatsby23,

CMake seems to be using python 2.7 just right.

It seems that that mono_tum is not loading the python module and that's the reason of the core dumped.

Can you please check if the paths are correct ?. Depending the directory you are running the program from (e.g. ./Examples/Monocular or ./ )  this may change (e.g. ../../src/python or ./src/python). Make sure that the path the program is using are the correct one.

Gatsby23, 2019-02-19:
OK! I really appreciate for your help! It can work now, but another question also bothering me that I can't see the point cloud? Do you publish the code about that ?

jmfacil, 2019-02-19:
Glad it helped! 

About the point cloud we did not released any code  #8 . What we did is: first estimatingall camera poses and second we constructed the point cloud offline using RGB-D images plus the dynamic objects mask that we generate. You can try that if you want dense reconstruction.

closed 2019-02-19 by jmfacil

Gatsby23, 2019-02-19:
OK， I'll Try it !

skydddive, 2019-03-12:
@Gatsby23 你怎么修改的啊，，我也遇到这个问题了，请问你修改的路径是在哪个文件里面的？

Chrislzy1993, 2019-03-18:
@Gatsby23 , 大兄弟,  问题解决了吗? 我也遇到了同样的问题

shuchun1997, 2020-09-21:
> OK! I really appreciate for your help! It can work now, but another question also bothering me that I can't see the point cloud? Have you publish the code about that ?
could you tell me how to fix the problem??
DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->py_module != NULL' failed