BertaBescos/DynaSLAM #13: about the opencv version?

skydddive, 2019-03-05:
hello ,your project is amazing.but when i build your project,it makes error.
Does your program have to use opencv 2.4.11?it is not be opencv 3.4.1?

BertaBescos, 2019-03-05:
Hi @skydddive! Thanks for the nice words. Are you using DynaSLAM's latest version? I made a few changes a couple of weeks ago so that it works for both OpenCV 2.4.11 and 3.4.1. Before that, only OpenCV 2.4.11 was supported.

If you have the latest version let me know and I will check this issue along the week.

skydddive, 2019-03-06:
hi,@BertaBescos .I use the latest version which you update,i get the error :
     /src/Conversion.cc:133:82 error:pure-specifier on function-definition 
           bool allocate(UMatData* U,int accessflags,UMatUsageFlags usageFlags) const=0{}
     /src/Conversion.cc:138:43 error:pure-specifier on function-definition ...
     /src/Conversion.cc:147:94 error:pure-specifier on function-definition ...
i solve this error by changing 'const = 0' to 'const'.

BertaBescos, 2019-03-06:
Hi @skydddive. That's actually a bug. Thanks for letting us know. I will fix it asap.

BertaBescos, 2019-03-06:
It is already fixed! I close the issue.

closed 2019-03-06 by BertaBescos