BertaBescos/DynaSLAM #14: about the "this->py_module "error

skydddive, 2019-03-07:
hello, @BertaBescos,i met the issue as the https://github.com/BertaBescos/DynaSLAM/issues/11#issue-411483365:
 /home/robotics/DynaSLAM/src/MaskNet.cc:39: DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->py_module != NULL' failed.
Aborted (core dumped)

but i can't understand the solution :
 "Depending the directory you are running the program from (e.g. ./Examples/Monocular or ./ ) this may change (e.g. ../../src/python or ./src/python). Make sure that the path the program is using is the correct one."
       I run the command in my computer:
./Example/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM3.yaml /home/skydddive/datasets/tum/rgb_freiburg3_walking_xyz ,
[where do i need to change "./src/python"?Could you please help me ..]

closed 2019-03-07 by skydddive

shuchun1997, 2020-09-21:
did you fix this problem?

Dongdong-NEU, 2022-11-27:
I also encountered the same problem:  Check.py,it runs successfully., however Importing Mask R-CNN Settings... Segmentation fault