BertaBescos/DynaSLAM #16: Bus error (core dumped)

kxhit, 2019-03-13:
Hi! I met an error when running the code. I have tried this both with cpu and gpu.
```
Initialated Mask RCNN network...
Creating net instance...
Loading net parameters...
Bus error (core dumped)
```
I run the Check.py successfully and can run the code without [PATH TO MASK] [PATH TO OUTPUT] parameters. Could anyone help me? Thanks!!!

fanguohao, 2019-03-16:
> Hi! I met an error when running the code. I have tried this both with cpu and gpu.
>
> ```
> Initialated Mask RCNN network...
> Creating net instance...
> Loading net parameters...
> Bus error (core dumped) [...]

Hello there! Is your problem solved? I also encountered the same problem.

leah0206, 2019-03-16:
@fanguohao @kxhit Hello! I also encountered the same problem. My configuration is GTX 1080. The version of my opencv is 3.4.1

leah0206, 2019-03-16:
@BertaBescos I am sorry to bother you. Do you have any suggestions on this problem?

jsdd25, 2019-03-16:
I have fixed this problem. Just change your opencv version to 2.4.11

kxhit, 2019-03-18:
@BertaBescos  I can run the code with openCV2 but not OpenCV3. I find the program not fluent either with MaskRCNN or not. Waiting for your reply. Thanks!

jmfacil, 2019-03-19:
I could not reproduce the problem. I would suggest to do as @jsdd25 proposes and change OpenCV version to 2.4.11.  We will write you back if we find out what the problem is.

fangzhangyun, 2019-03-28:
I also have this problem. I find the C++ can't call the MaskRCNN.py.
DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->py_module != NULL' failed.
I don't know how to solve it.  Can you help me?

BertaBescos, 2019-03-28:
Are you using OpenCV 2.4.11? If not, I would recommend you to use it. 
We will try to fix this issue for newer versions within the next weeks.

fangzhangyun, 2019-03-28:
> Are you using OpenCV 2.4.11? If not, I would recommend you to use it.
> We will try to fix this issue for newer versions within the next weeks.

Thanks for your answer,I had changed my OpenCV from 2.4.13 to 2.4.11 ,but  it is useless, I am looking forward to your newer versions

BertaBescos, 2019-03-28:
Then the problem must be something else. It should work without problems on OpenCV 2.4.11.
Are you running it from the DynaSLAM directory" path?

fangzhangyun, 2019-03-28:
> Then the problem must be something else. It should work without problems on OpenCV 2.4.11.
> Are you running it from the DynaSLAM directory path?

I run it from DynaSLAM directory path like the ORB_SLAM2,it succeeded. when I added the parameter [PATH_TO_MASK] [PATH_TO_OUTPUT] to call the MaskRCNN, the problem has happened.

fangzhangyun, 2019-03-28:
> Then the problem must be something else. It should work without problems on OpenCV 2.4.11.
> Are you running it from the DynaSLAM directory path?

I found the python version is the key problem. Mask_RCNN requires the Python3.4 + TensorFlow1.3 + Keras2.0.8,but your DynaSLAM is python2.7. I will try this way to build the DynaSLAM.

PushyamiKaveti, 2019-06-25:
Any luck with this error? I have the same problem

qxdaaaaa, 2019-06-26:
I have meet the trouble for several days.But ,i can't solver it .my system is Ubuntu16.04+cuda8.0+cudnn6+opencv 2.4.11+orbslam2 environment.Can anyone help me ?

PushyamiKaveti, 2019-06-26:
I got it working finally. The issue on my side was opencv 3.3. It works with opencv 2.4.10. When I debugged I found that the problem is in Conversion.cc inside the method toNDArray(const cv::Mat& m). The method has two sections for two opencv versions (2 and 3) in case of opencv 3 the PyObject returned from Mat seems to be the problem when it gets passed to GetDynSeg() of  MaskRCNN.py. I am trying to fix it now.

qxdaaaaa, 2019-06-27:
You are too good！Can you tell me  about your configuration environment in detail ?and have you run check.py？I have trouble again yesterday, maybe I have to configure the environment for this code again.

qxdaaaaa, 2019-06-27:
@PushyamiKaveti Can you help me ?

qxdaaaaa, 2019-06-28:
@BertaBescos  Can you help me? When I give [PATH TO MASK] [PATH TO OUTPUT] ,It is "core dumped".

qxdaaaaa, 2019-06-28:
@PushyamiKaveti @BertaBescos My environment is :Ubuntu16.04 +cuda8.0+cudnn6+opencv2.4.11+python2.7+orbslam2.

qxdaaaaa, 2019-06-30:
I have solved this problem, thank you for your help.@PushyamiKaveti @BertaBescos

jzpwhu1996, 2019-11-14:
> I have solved this problem, thank you for your help.@PushyamiKaveti @BertaBescos

I meet the same problem, can you help me?
My error is Bus error too,  my python is 2.7, opencv version is 2.4.11,and Ican run Check.py successfully.
@qxdaaaaa

zxGH86, 2019-12-08:
I meet the same problem, can you help me?My error is Bus error too, my python is 2.7, opencv version is 2.4.11 and 3.4.1,and Ican run Check.py successfully. @kxhit

zxGH86, 2019-12-08:
I meet the same problem, can you help me?
My error is Bus error too, my python is 2.7, opencv version is 2.4.11 and 3.4.1,tensorflow-gpu1.4.0 and I can run Check.py successfully.
@qxdaaaaa @kxhit @fanguohao

qxdaaaaa, 2019-12-08:
you can test the tensflow .the problem ,may be the tensflow.you can use tf==cpu------------------ 原始邮件 ------------------

 
&mdash;

zxGH86, 2019-12-09:
I just test the masknet .cc find that the     cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
....  PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image);   ......}  doesn't working, how to solve this problem?
@qxdaaaaa

zxGH86, 2019-12-09:
Thanks,@qxdaaaaa.    I just test the masknet .cc find that the function cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......} doesn't working, how to solve this problem?

zxGH86, 2019-12-09:
Thanks,@qxdaaaaa. I just test the masknet .cc find that the function```
 cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......} 
```doesn't working, how to solve this problem?

zxGH86, 2019-12-09:
Thanks,@qxdaaaaa. I just test the masknet .cc find that the function```
 cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......}
``` doesn't working, how to solve this problem?

zxGH86, 2019-12-09:
Thanks,@qxdaaaaa.``` I just test the masknet .cc find that the function cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......} ```doesn't working, how to solve this problem?

zxGH86, 2019-12-09:
Thanks,@qxdaaaaa. I just test the masknet .cc find that the function``` cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......}``` doesn't working, how to solve this problem?

SuperBigBrother, 2020-03-06:
> I meet the same problem, can you help me? My error is Bus error too, my python is 2.7, opencv version is 2.4.11 and 3.4.1,tensorflow-gpu1.4.0 and I can run Check.py successfully.

I‘d like to ask if cuda10.0 + tensorflow1.13.1 is ok? I think I have correctly installed with  python2 but I can't pass check when I use terminal to run 'python check.py'. I think I need you help.....

shubhendu1907, 2020-06-27:
> I have solved this problem, thank you for your help.@PushyamiKaveti @BertaBescos

How did you solve the problem?
My system is Python 2.7, OpenCV 3.3.0 , tensorflow-gpu=1.13.1, CUDA 10.0 and I can run Check.py successfully.

KaninchenM, 2020-10-15:
> > I have solved this problem, thank you for your help.@PushyamiKaveti @BertaBescos
> 
> How did you solve the problem? @PushyamiKaveti @qxdaaaaa
> My system is Python 2.7, OpenCV 3.3.0 , tensorflow-gpu=1.13.1, CUDA 10.0 and I can run Check.py successfully.

Have you run successfully in cuda10?

hanxiumeng, 2021-01-18:
> I got it working finally. The issue on my side was opencv 3.3. It works with opencv 2.4.10. When I debugged I found that the problem is in Conversion.cc inside the method toNDArray(const cv::Mat& [...]

Yes,it is the problem,the Conversion.cc is failed with opencv3.And I have fixed it, here is my Conversion.cc:
------------------------------------------------------------------------------------------------------------------------------------------------------
#include "Conversion.h"
#include <iostream>

namespace DynaSLAM
{

static void init()
{
    import_array();
}

static PyObject* failmsgp(const char *fmt, ...)
{
    char str[1000];

    va_list ap;
    va_start(ap, fmt);
    vsnprintf(str, sizeof(str), fmt, ap);
    va_end(ap);

    PyErr_SetString(PyExc_TypeError, str);
    return 0;
}

using namespace cv;
//===================   ERROR HANDLING     =========================================================

static int failmsg(const char *fmt, ...) {
	char str[1000];

	va_list ap;
	va_start(ap, fmt);
	vsnprintf(str, sizeof(str), fmt, ap);
	va_end(ap);

	PyErr_SetString(PyExc_TypeError, str);
	return 0;
}

//===================   THREADING     ==============================================================
class PyAllowThreads {
public:
	PyAllowThreads() :
			_state(PyEval_SaveThread()) {
	}
	~PyAllowThreads() {
		PyEval_RestoreThread(_state);
	}
private:
	PyThreadState* _state;
};

class PyEnsureGIL {
public:
	PyEnsureGIL() :
			_state(PyGILState_Ensure()) {
	}
	~PyEnsureGIL() {
		PyGILState_Release(_state);
	}
private:
	PyGILState_STATE _state;
};

enum {
	ARG_NONE = 0, ARG_MAT = 1, ARG_SCALAR = 2
[... 196 lines cut ...]
	return m;

}

PyObject* NDArrayConverter::toNDArray(const cv::Mat& m)
{
    if (!m.data)
		Py_RETURN_NONE;
		Mat temp,
	*p = (Mat*) &m;
	if (!p->u || p->allocator != &g_numpyAllocator) {
		temp.allocator = &g_numpyAllocator;
		ERRWRAP2(m.copyTo(temp));
		p = &temp;
	}
    std::cout<<p->u->refcount<<std::endl;
    std::cout<<p->u->userdata<<std::endl;
	PyObject* o = (PyObject*) p->u->userdata;
	Py_INCREF(o);
	return o;

}

}
-----------------------------------------------------------------------------------------------------------------------

lijingyu16, 2021-04-08:
> 
> 
> > I got it working finally. The issue on my side was opencv 3.3. It works with opencv 2.4.10. When I debugged I found that the problem is in Conversion.cc inside the method toNDArray(const cv::Mat& m). The method has two sections for two opencv versions (2 and 3) in case of opencv 3 the PyObject returned from Mat seems to be the problem when it gets passed to GetDynSeg() of MaskRCNN.py. I am trying to fix it now.
> 
> ## Yes,it is the problem,the Conversion.cc failed with opencv3.And I have fixed it, here is my Conversion.cc:
> 
> #include "Conversion.h"
> #include
> 
> namespace DynaSLAM
> {
> 
> static void init()
> {
> import_array();
> }
> 
> static PyObject* failmsgp(const char *fmt, ...)
> {
> char str[1000];
> 
> ```
> va_list ap;
> va_start(ap, fmt);
> vsnprintf(str, sizeof(str), fmt, ap);
> va_end(ap);
> 
> PyErr_SetString(PyExc_TypeError, str);
> return 0;
> ```
> 
> }
> 
> using namespace cv;
> //=================== ERROR HANDLING =========================================================
> 
> static int failmsg(const char *fmt, ...) {
> char str[1000];
> 
> ```
> va_list ap;
> va_start(ap, fmt);
> vsnprintf(str, sizeof(str), fmt, ap);
> va_end(ap);
> 
> PyErr_SetString(PyExc_TypeError, str);
> return 0;
> ```
> 
> }
> 
> //=================== THREADING ==============================================================
> class PyAllowThreads {
> public:
> PyAllowThreads() :
> _state(PyEval_SaveThread()) {
> }
> ~PyAllowThreads() {
> PyEval_RestoreThread(_state);
> }
> private:
> PyThreadState* _state;
> };
> 
> class PyEnsureGIL {
> public:
> PyEnsureGIL() :
> _state(PyGILState_Ensure()) {
> }
> ~PyEnsureGIL() {
[... 202 lines cut ...]
> ```
> 
> }
> 
> PyObject* NDArrayConverter::toNDArray(const cv::Mat& m)
> {
> if (!m.data)
> Py_RETURN_NONE;
> Mat temp,
> _p = (Mat_) &m;
> if (!p->u || p->allocator != &g_numpyAllocator) {
> temp.allocator = &g_numpyAllocator;
> ERRWRAP2(m.copyTo(temp));
> p = &temp;
> }
> std::cout<u->refcount<<std::endl;
> std::cout<u->userdata<<std::endl;
> PyObject* o = (PyObject*) p->u->userdata;
> Py_INCREF(o);
> return o;
> 
> }
> ## }

Hello, I can run DynaSLAM successfully with OPENCV2.4.11, but after using OPENCV3.4.9, there is always report a bus error, a core dump; after seeing the Conversion.cc  which you changed, While using it to Compile DynaSLAM ，it also report some error, so，I want to know how do you run DNASLAM using opencv3?  Thank you very much, hope you can contact my email：joyfourier@163.com , looking forward to your reply!

lijingyu16, 2021-04-08:
@hanxiumeng   Thanks to your work, I was able to run successfully under OPENCV2.4.11. There were bus errors when using OPENCV3.4.9, and many errors when compiling later when using the files（Conversion.cc）  you modified, i want to know how you successfully run this project under OPENCV3, and I look forward to your help：joyfourier@163.com

mobiiin, 2021-07-15:
> ## Yes,it is the problem,the Conversion.cc failed with opencv3.And I have fixed it, here is my Conversion.cc:
> #include "Conversion.h"
> #include
>
> namespace DynaSLAM
> {
>
> static void init()
> { [...]

your code doesn't work either, leads to the same error
could you share the modified conversion.cc file? did you get it to work with opencv 3?