BertaBescos/DynaSLAM #18: cannot declare variable ‘DynaSLAM::g_numpyAllocator’

Heredx, 2019-03-18:
`/home/xh/Projects/SLAM/DynaSLAM/src/Conversion.cc:205:16: error: cannot declare variable ‘DynaSLAM::g_numpyAllocator’ to be of abstract type ‘DynaSLAM::NumpyAllocator’`

`NumpyAllocator g_numpyAllocator;`

`/home/xh/Projects/SLAM/DynaSLAM/src/Conversion.cc:74:7: note:   because the following virtual functions are pure within ‘DynaSLAM::NumpyAllocator’:`

`class NumpyAllocator : public MatAllocator
In file included from /usr/local/include/opencv2/core.hpp:59:0,
from/usr/local/include/opencv2/imgproc.hpp:46,
from/home/xh/Projects/SLAM/DynaSLAM/include/Conversion.h:14,
from/home/xh/Projects/SLAM/DynaSLAM/src/Conversion.cc:9:`

I got this issue when I run `make -j12` in `~/Projects/SLAM/DynaSLAM/build`.
But I don't find any pure virtual functions within ‘DynaSLAM::NumpyAllocator’.
Do you know why this occurs? Looking forward to your answer.

jmfacil, 2019-03-18:
It seems like an OpenCV version issue. Make sure you are using last version of DynaSLAM #13

Heredx, 2019-03-19:
Thanks. I've solved this problem.
While I meet another issue when I run
`./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM2.yaml data/rgbd_dataset_freiburg2_desk_with_person .//Examples/RGB-D/associations.txt ./maskfolder ./output`
The terminal shows:
`nImages: 3983`
`Loading Mask R-CNN. This could take a while...`
`Importing Mask R-CNN Settings...`
`Segmentation fault (core dumped)`
Looking forward to your answer.

Heredx, 2019-03-19:
What's more, the program can run well without using MaskRCNN
`./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM2.yaml data/rgbd_dataset_freiburg2_desk_with_person .//Examples/RGB-D/associations.txt`

jmfacil, 2019-03-20:
Hi! What versions of Python and OpenCV are you using? Check #19. Maybe it helps.

Heredx, 2019-03-20:
I've solved it. Thank you so much!

jmfacil, 2019-03-20:
Was it a version problem?

Heredx, 2019-03-21:
It was because I altered CmakeLists.txt and changed python directory to use python of anaconda, while some mistakes happened. Then I used python of Linux to cmake&make the program, it could compile successfully.

jmfacil, 2019-03-21:
That's great :). I will close the thread then.

closed 2019-03-21 by jmfacil

Heredx, 2019-03-21:
Thanks for helping.

ashsifat, 2022-03-29:
Hi, I am having the exact same issue, I didn't change the CmakeLists.txt to anything other than use opencv 4.1.1. Any idea what the problem might be in my case? downgrading to opencv3.2 is not an option since jetpack for nvidia xavier agx only allows me to properly install opencv4 now.

/home/nvidia/YOLO-DynaSLAM/src/Conversion.cc:206:16: error: cannot declare variable ‘DynaSLAM::g_numpyAllocator’ to be of abstract type ‘DynaSLAM::NumpyAllocator’
  206 | NumpyAllocator g_numpyAllocator;