BertaBescos/DynaSLAM #19: run error about r-cnn

Chrislzy1993, 2019-03-18:
hi, every big god! 
when i run the code with R-CNN, it show the following error
```
nImages: 795
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
[1]    29733 segmentation fault (core dumped)  ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml
```
and when i run Check.py , it show the following error
```
Traceback (most recent call last):
  File "Check.py", line 6, in <module>
    import skimage.io
ImportError: No module named skimage.io
```
python version is: 2.7.12, opencv version is: 3.3.1

kxhit, 2019-03-18:
You can install skimage with pip.

Chrislzy1993, 2019-03-18:
@kxhit , thanks for your answer,  after you suggestion,  i run Check.py file correctly, but when i run dyna-slam, it show the same segment fault too
```
(python36) ➜  python git:(master) ✗ python Check.py 
Using TensorFlow backend.
Initializing Mask RCNN network...
./
2019-03-18 17:01:23.593842: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 FMA
2019-03-18 17:01:23.598244: I tensorflow/core/common_runtime/process_util.cc:69] Creating new thread pool with default inter op setting: 2. Tune using inter_op_parallelism_threads for best performance.
Initialated Mask RCNN network...
Mask R-CNN is correctly working
```
but when i run dyna-slam 
```
(python36) ➜  DynaSLAM git:(master) ✗ ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml /media/liuzhiyang/data/dataset-tum/TUM3/rgbd_dataset_freiburg3_sitting_rpy Examples/RGB-D/associations/associations.txt src/python/mask_rcnn_coco.h5 data
nImages: 2198
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
[1]    12593 segmentation fault (core dumped)  ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml    
```

Chrislzy1993, 2019-03-18:
gdb:
```
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
[New Thread 0x7fffcfe00700 (LWP 13197)]
[New Thread 0x7fffcf5ff700 (LWP 13198)]
[New Thread 0x7fffcadfe700 (LWP 13199)]

Thread 1 "rgbd_tum" received signal SIGSEGV, Segmentation fault.
0x00007ffff1e261c0 in PyObject_GetAttrString ()
   from /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0
```

Chrislzy1993, 2019-03-18:
i use anaconda to create a new env, and python version is 3.6, but why here still get python2.7.so??? confused

jmfacil, 2019-03-18:
Current version of the code runs with Python 2.7 [(lline in the code)](https://github.com/BertaBescos/DynaSLAM/blob/7dd59efcf463095279f6567c2473c80dd9c48991/include/MaskNet.h#L15) , we plan to update it to 3.X but the code is not fully tested yet.

It does not matter if you work on a virtual env with python3.x the code seeks for python 2.7 specifically.

Chrislzy1993, 2019-03-18:
@jmfacil , hi, jmfacil, 
thanks for your replaying,  i understand use libpython2.7.so in cmakelists.txt, and code can find it,  but my code still run error with segment fault??  and i can run orb-slam2 code correctly before when i use opencv 3.3.1, i am confused about where wrong happened?

jmfacil, 2019-03-18:
Hi there,
Do you get same segment fault? 
First, make sure you are using last version of DynaSLAM #13.
Second make sure all dependencies are correctly installed for python2.7

Chrislzy1993, 2019-03-19:
@jmfacil , yes, i git clone the last version of DynaSLAM , about python 2.7, it is installed along with system,  so i don't konw the meaning of dependencies for python2.7.  cause using mask-rcnn,  i install python 3.6 in a new env by anaconda, and installed tensorflow and keras like readme says, and i can run check.py correctly, thanks again for you nice reply!!!

jmfacil, 2019-03-19:
Hi! No problem!
Dependencies like numpy, tensorflow ... should be installed on python2.7 as is the version DynaSLAM will be using. Mask-rcnn will be executed by the program with python2.7. I would recommend you create another env but this time with python 2.7. And then install dependencies there. I didn't try with virtual env yet so maybe you need to make sure paths to python libs are correct. Check CMakeList.txt 
Another option would be to install dependencies on the python 2.7 installed with the system. Then changes in cmake and code should be minimal. 

Hope it helps!

Chrislzy1993, 2019-03-19:
@jmfacil , thanks , i change my python version into 2.7 follow your advice,  and i install dependencies needed by mask-rcnn.  i run check.py,  it result correct.  and i compile the dynaslam,  build sucessfully. but when i run code, it show about bus error,  different from before error:
```
➜  DynaSLAM git:(master) ✗ ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml /media/liuzhiyang/data/dataset-tum/TUM3/rgbd_dataset_freiburg3_sitting_rpy Examples/RGB-D/associations/fr3_sitting_rpy.txt src/python/mask_rcnn_coco.h5 data
nImages: 795
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
/usr/local/lib/python2.7/dist-packages/pycocotools/coco.py:49: UserWarning: 
This call to matplotlib.use() has no effect because the backend has already
been chosen; matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
or matplotlib.backends is imported for the first time.

The backend was *originally* set to u'TkAgg' by the following code:
  File "/home/liuzhiyang/cat/DynaSLAM/src/python/MaskRCNN.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 72, in <module>
    from matplotlib.backends import pylab_setup
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/__init__.py", line 14, in <module>
    line for line in traceback.format_stack()

  import matplotlib; matplotlib.use('Agg')
Using TensorFlow backend.
Initializing Mask RCNN network...
./src/python
2019-03-19 20:15:34.410090: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2019-03-19 20:15:34.503698: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:964] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2019-03-19 20:15:34.504107: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1432] Found device 0 with properties: 
name: GeForce GTX 1060 6GB major: 6 minor: 1 memoryClockRate(GHz): 1.7845
pciBusID: 0000:01:00.0
totalMemory: 5.93GiB freeMemory: 5.23GiB
2019-03-19 20:15:34.504124: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1511] Adding visible gpu devices: 0
2019-03-19 20:15:34.730313: I tensorflow/core/common_runtime/gpu/gpu_device.cc:982] Device interconnect StreamExecutor with strength 1 edge matrix:
2019-03-19 20:15:34.730347: I tensorflow/core/common_runtime/gpu/gpu_device.cc:988]      0 
2019-03-19 20:15:34.730367: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1001] 0:   N 
2019-03-19 20:15:34.730550: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1115] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 4990 MB memory) -> physical GPU (device: 0, name: GeForce GTX 1060 6GB, pci bus id: 0000:01:00.0, compute capability: 6.1)
Initialated Mask RCNN network...
Creating net instance...
Loading net parameters...
[1]    30426 bus error (core dumped)  ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml
```
did u meet this error before????, thanks again for your careful advise

jmfacil, 2019-03-19:
In #16 they suggest to change to OpenCV 2.4.11 instead of 3. Maybe you can try that.

Chrislzy1993, 2019-03-20:
@jmfacil , thanks,  i am trying,  and there is some problem when  opencv 2.4.11 with cuda 9,

Chrislzy1993, 2019-03-20:
@jmfacil   thanks for your always nice reply ,  and the code can run  when i change opencv version from 3.3.1 to 2.4.11,  even there are some problem with opencv and cuda. 
[image]

closed 2019-03-20 by Chrislzy1993

jmfacil, 2019-03-20:
@Chrislzy1993 awesome!! I am glad it worked :)