BertaBescos/DynaSLAM #20: about mono style

Chrislzy1993, 2019-03-20:
in mono style, i find there are two difficulty
1. it is too hard to initialize
[image]
2. it is too easy to lose localization
[image]

in question #12 , advise to uprise the detect num,  i try it , but has nothing different. its anything i can done to solve this problem, like imu ????

Chrislzy1993, 2019-03-20:
@jmfacil @BertaBescos , thanks a lot

jmfacil, 2019-03-20:
Initialization when using monocular is usually more difficult of course. It depends on the sequence (i.e. the parallax, the features, dynamic objects....) including more sensors like infrared cameras (RGB-D), Stereo and/or IMU would help if fuse correctly for sure. But we are not planning to include IMU integration.

Chrislzy1993, 2019-03-25:
enen, thanks a lot , when i test tum other dateset , like rgbd_dataset_freiburg3_walking_rpy,  it can work well, may be the problem is in dataset.

closed 2019-03-25 by Chrislzy1993