BertaBescos/DynaSLAM #23: build error about -- no member named ‘mvpMatchedLoop’

SUNLEYUAN, 2019-04-09:
--------------------------------------------------------------------------------------------
[  6%] Building CXX object CMakeFiles/DynaSLAM.dir/src/System.cc.o
/home/sunleyuan/DynaSLAM/src/System.cc: In member function ‘void ORB_SLAM2::System::SaveDetectedLoops(const string&)’:
/home/sunleyuan/DynaSLAM/src/System.cc:467:72: error: ‘class ORB_SLAM2::LoopClosing’ has no member named ‘mvpMatchedLoop’
  vector<pair<KeyFrame*, KeyFrame*> > mvpMatchedLoop = mpLoopCloser->mvpMatchedL
                                                                     ^
CMakeFiles/DynaSLAM.dir/build.make:62: recipe for target 'CMakeFiles/DynaSLAM.dir/src/System.cc.o' failed
--------------------------------------------------------------------------------------------
Please give me some guidance, thanks!

BertaBescos, 2019-04-09:
Sorry, that was actually a bug! It should be solved now.

SUNLEYUAN, 2019-04-09:
> Sorry, that was actually a bug! It should be solved now.

Have already been solved?  I have used the latest version of your code. This error has crushed the build process totally.
How to solve the error?
Thanks!

BertaBescos, 2019-04-09:
I have just fixed the bug, right after seeing your issue. I had not noticed it before.

SUNLEYUAN, 2019-04-09:
> I have just fixed the bug, right after seeing your issue. I had not noticed it before.

Sorry, I have noticed your code update, I will try.
 Anyway, thanks for your reply in time!

SUNLEYUAN, 2019-04-09:
checked, already been fixed!

closed 2019-04-09 by SUNLEYUAN