BertaBescos/DynaSLAM #24: Importing Mask R-CNN Settings...   Segmentation fault (core dumped)

SUNLEYUAN, 2019-04-09:
The build process could be finished.
But under the Mask R-CNN detecting model, the Segment fault occurred.
------
:~/DynaSLAM-master$ ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml rgbd_dataset_freiburg3_walking_xyz/ associate.txt mask/ output/
nImages: 827
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
Segmentation fault (core dumped)
------

After that I checked the Check.py file for Mask R-CNN, the result as bellowed:
------
~/DynaSLAM-master/src/python$ python Check.py 
Traceback (most recent call last):
  File "Check.py", line 10, in <module>
    import coco
  File "/home/sunleyuan/DynaSLAM-master/src/python/coco.py", line 40, in <module>
    from pycocotools.coco import COCO
  File "/usr/local/lib/python2.7/dist-packages/pycocotools/coco.py", line 55, in <module>
    from . import mask as maskUtils
  File "/usr/local/lib/python2.7/dist-packages/pycocotools/mask.py", line 3, in <module>
    import pycocotools._mask as _mask
  File "__init__.pxd", line 918, in init pycocotools._mask
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject
------

plz, give me some instructions.
Sorry for open an issue again.

BertaBescos, 2019-04-09:
Run it from the DynaSLAM folder as ```python src/python/Check.py```.

SUNLEYUAN, 2019-04-09:
> Run it from the DynaSLAM folder as `python src/python/Check.py`.

I think it is the same with : ~/DynaSLAM-master/src/python$ python Check.py

The results are the same actually,
------
~/DynaSLAM-master$ python src/python/Check.py 
Traceback (most recent call last):
  File "src/python/Check.py", line 10, in <module>
    import coco
  File "/home/sunleyuan/DynaSLAM-master/src/python/coco.py", line 40, in <module>
    from pycocotools.coco import COCO
  File "/usr/local/lib/python2.7/dist-packages/pycocotools/coco.py", line 55, in <module>
    from . import mask as maskUtils
  File "/usr/local/lib/python2.7/dist-packages/pycocotools/mask.py", line 3, in <module>
    import pycocotools._mask as _mask
  File "__init__.pxd", line 918, in init pycocotools._mask
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject
------

BertaBescos, 2019-04-09:
It looks like it is a numpy version problem.

SUNLEYUAN, 2019-04-09:
I have checked the closed issue about the version of numpy, you said that your version is 1.15.2, so I have used the same version with you. So maybe it is not the version problem.
---
numpy                              1.15.2     
---

BertaBescos, 2019-04-09:
I am afraid I can not really help you out with this. It seems to be more of a Mask R-CNN installation issue.
Maybe this issue can help (https://github.com/cocodataset/cocoapi/issues/248).

SUNLEYUAN, 2019-04-10:
> I am afraid I can not really help you out with this. It seems to be more of a Mask R-CNN installation issue.
> Maybe this issue can help ([cocodataset/cocoapi#248](https://github.com/cocodataset/cocoapi/issues/248)).

Thanks, I have used this issue's solution:
---
I solve it like #14 (comment)

$ git clone https://github.com/pdollar/coco.git
$ cd coco/PythonAPI
$ make
$python setup.py install

and change the numpy version to 1.15.1
---

Now, the Check.py result is :
---
Initialated Mask RCNN network...
Mask R-CNN is correctly working
---

closed 2019-04-10 by SUNLEYUAN