BertaBescos/DynaSLAM #28: how to evaluate the kitti dateset with rpe rre and ate as your paper

qiqzhang, 2019-05-14:
could you supply the evaluate code, or tell how to evaluate  rpe rre and ate. looking forward to u reply

BertaBescos, 2019-05-14:
Hi @qiqzhang ! I used the evaluation tools provided by the KITTI authors in their website (http://www.cvlibs.net/datasets/kitti/eval_odometry.php) under the "Download odometry development kit (1 MB)" folder.

qiqzhang, 2019-05-17:
Thanks for your answer. When I try to get the results, I got an error about mail->finalize. Then
I delete the code `  if (argc==4) mail->finalize(success,"odometry",result_sha,argv[2]);
  else         mail->finalize(success,"odometry",result_sha);`
and input with two parameter in terminal，got  *rl* 、*rs*、*tl* 、*ts*...，but the rusult as fellow， I dont know why the result is strange and how did you do with code, or something else?
[image]

qiqzhang, 2019-05-18:
how to calculate the result as your paper, online waiting...

closed 2019-05-21 by qiqzhang

linpeisensh, 2021-03-03:
@qiqzhang @BertaBescos 
Could you please tell me how to use odometry development kit to get the rre, rte and rpe? I compiled and run the evaluate_odometry and only got the error file, plot error, plot path and stats.txt.
Thank you!