BertaBescos/DynaSLAM #35: Light Tracking not working because Tracking is not initialized... ？

fsxss, 2019-10-25:
why the error will cause ? if  you have solved this problem ,please tell me Thanks !!

BertaBescos, 2019-10-25:
That happens when tracking is not initialized. Unless the scene is extremely dynamic in the beginning it should initialize within the first frame. (I assume you are using the RGB-D setup.)

fsxss, 2019-10-25:
> That happens when tracking is not initialized. Unless the scene is extremely dynamic in the beginning it should initialize within the first frame. (I assume you are using the RGB-D setup.)

Thanks  for your reply ! yes , I using the RGB-D setup and running the code in TUM datasets( fr3_s _rpy and fr3_w_rpy). but it can't be initialized in any datasets, 
so I increased the number of extracted features per frame in the corresponding settings file TUM3.yaml from 1000 to 1500 as you said befor. the system can be initialized.
however, new question was occuring , after tracking few frames ,the code dead and showed "Segmentation Fault"

hperik1, 2020-03-03:
I am experiencing that Light tracking is not working on fr3_w_xyz. Even with the update from 1000 to 1500. Any further experiences?

KaninchenM, 2020-06-15:
> I am experiencing that Light tracking is not working on fr3_w_xyz. Even with the update from 1000 to 1500. Any further experiences?

I am also using  fr3_w_xyz, setting the number of extracted features as 3000 makes no use for me. Did you solve the problem?

liubamboo, 2021-04-07:
@BertaBescos 
hello, when I used the RGB-D setup and running the code in TUM datasets(fr3_walking_halfsphere), default config. I saw the same problem: Light Tracking not working because Tracking is not initialized... Geometry not working.

and then I change the ORBextractor.nFeatures to 1500 2000 2500 3000. but the problem still exist. Can you tell me which config you used? I saw there a good result in the paper.
[image]

Please help me, Thanks a lot.

JeonHyeongJunKW, 2022-04-12:
> > That happens when tracking is not initialized. Unless the scene is extremely dynamic in the beginning it should initialize within the first frame. (I assume you are using the RGB-D setup.)
> 
> Thanks for your reply ! yes , I used the RGB-D setup and running the code in TUM datasets( fr3_s _rpy and fr3_w_rpy). but it can't be initialized in any datasets, so I increased the number of extracted features per frame in the corresponding settings file TUM3.yaml from 1000 to 1500 as you said before. the system can be initialized.Thank you!!

good!