BertaBescos/DynaSLAM #37: getting seg-faults when importing mask r-cnn

mshong0320, 2019-11-07:
I ran the command to run dyna slam for mono_kitti adding at the end the path to the mask rcnn, which for me was at "./src/python/mask_rcnn_coco.h5", but I am getting seg faults.

Anyone with solution to this?

Thank you in advance!

yiran1012, 2019-11-08:
I have also meet this problem, but I haven't find out the solution.

mshong0320, 2019-11-08:
searched through all of the closed issues addressing the same problem, but not much help found.

In the MaskNet.cc the line for PyImport_ImportModule("MaskRCNN") is returning NULL and I've set the PySys_SetPath to ./src/python/ and setenv("PYTHONPATH", x) is set to ./src/python as well. 

Any suggestions?

jzpwhu1996, 2019-11-14:
May be you can try to open the src/python/Maskrcnn.py and change the ./src/python/ to an absolute path, such as /home/xxx/DynaSLAM/src/python

zxGH86, 2019-12-09:
I just test the masknet .cc find that the function```
 cv::Mat SegmentDynObject::GetSegmentation(cv::Mat &image,std::string dir, std::string name){
.... PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image); ......}
``` doesn't working, how to solve this problem?``

fsxss, 2019-12-26:
which version of  opencv you used?  I  changed the opencv's version to 2.4.11 and this problem can be solved.

zxGH86, 2019-12-26:
I have  both opencv 2.4.11 and 3.4.1 ,because when I compile the program
it reports that I don't have the libopencv3.4.1.so, is that the reason why the function doesn't work?

zxGH86, 2019-12-26:
> 
> 
> which version of opencv you used? I changed the opencv's version to 2.4.11 and this problem can be solved.

zxGH86, 2019-12-26:
> 
> 
> which version of opencv you used? I changed the opencv's version to 2.4.11 and this problem can be solved.

I have both opencv 2.4.11 and 3.4.1, because when I compile the program ,it reports that I don't have
the libopencv3.4.1.so, is that the reason  the function` cv::Mat SegmentDynObject::GetSegmentation`
doesn't work? @fsxss

fsxss, 2019-12-26:
> I have both opencv 2.4.11 and 3.4.1, because when I compile the program ,it reports that I don't have
> the libopencv3.4.1.so, is that the reason the function` cv::Mat [...]

it probably is . whatever ,you should let the system find the libopencv*.so, make sure you have linked opencv director correctly  in CMakeLists.txt. I recommend use opencv 2.4.11，because when i used opencv 3 in my computer it can't work .

zxGH86, 2019-12-26:
> 
> it probably is . whatever ,you should let the system find the libopencv*.so, make sure you have linked opencv director correctly in CMakeLists.txt. I recommend use opencv 2.4.11，because when i used opencv 3 in my computer it can't work .
OK, thank you for your reply, I'll try it then to see what will happen. @fsxss

zxGH86, 2019-12-26:
> it probably is . whatever ,you should let the system find the libopencv*.so, make sure you have linked opencv director correctly in CMakeLists.txt. I recommend use opencv 2.4.11，because when i used [...]

OK, thank you for your reply, I'll try it then to see what will happen. @fsxss

zxGH86, 2019-12-26:
> which version of opencv you used? I changed the opencv's version to 2.4.11 and this problem can be solved.
I  have change the version of opencv to 2.4.11, but when run the system with rgbd_dataset_freiburg3_walking_xyz, it comes with `Light Tracking not working because Tracking is not initialized...` ,even if I  change the `ORBextractor.nFeatures` to 1500 or 2500 even 4000, how can
I solve this problem? @fsxss

fsxss, 2019-12-27:
> which version of opencv you used? I changed the opencv's version to 2.4.11 and this problem can be solved.
> I  have change the version of opencv to 2.4.11, but when run the system with [...]
I also met the not initialized  problem , but  I  solved this by change the KPs  to 1500 .you can try  run only in CNN case to find what's the problem cause .

zxGH86, 2019-12-27:
> which version of opencv you used? I changed the opencv's version to 2.4.11 and this problem can be solved.
> I  have change the version of opencv to 2.4.11, but when run the system with [...]

Thank you for your reply, I have solved the problem by removing the opencv 3.4.1 and increase the 
`ORBextractor.nFeatures `to 2000, then it can work correctly.@fsxss

pean1128, 2023-02-17:
In my cases, I repair some indent mistake in '*.py' files