BertaBescos/DynaSLAM #38: Importing Mask R-CNN Settings...Segmentation fault (core dumped)

jzpwhu1996, 2019-11-14:
Sorry to bothering you.
When I run the DynaSLAM with the command "./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml Data/rgbd_dataset_freiburg3_walking_xyz/ Data/rgbd_dataset_freiburg3_walking_xyz/associate.txt Data/Mask/ Data/Output/"

My terminal's output is :
nImages: 827
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
Segmentation fault (core dumped)

And I find the interrupt occurs at the MaskNet.cc "this->py_class = PyObject_GetAttrString(this->py_module, this->class_name.c_str());"
And I also find after the "this->py_module = PyImport_ImportModule(this->module_name.c_str());" been called, the py_module is NULL.

Could you help me to solve the problem?

jzpwhu1996, 2019-11-14:
According to a closed issue,I run the check.py ,the result is :
Using TensorFlow backend.
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Initializing Mask RCNN network...
./src/python
Traceback (most recent call last):
  File "Check.py", line 40, in <module>
    model.load_weights(COCO_MODEL_PATH, by_name=True)
  File "/home/jzp/DynaSLAM-master/src/python/model.py", line 1900, in load_weights
    f = h5py.File(filepath, mode='r')
  File "/usr/local/lib/python2.7/dist-packages/h5py/_hl/files.py", line 408, in __init__
    swmr=swmr)
  File "/usr/local/lib/python2.7/dist-packages/h5py/_hl/files.py", line 173, in make_fid
    fid = h5f.open(name, flags, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 88, in h5py.h5f.open
IOError: Unable to open file (unable to open file: name = './src/python/mask_rcnn_coco.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)

It seems that there is no mask_rcnn_coco.h5. But I really place the mask_rcnn_coco.h5 in the src/Python

jzpwhu1996, 2019-11-14:
Now，I can run Check.py sucessfully.The output is :Mask R-CNN is correctly working
And according to the closed issue,I changed my opencv version to 2.4.11 and rebuilt the DynaSLAM.
But when I run the TUM rgbd example,the output is:
Initializing Mask RCNN network...
/home/jzp/DynaSLAM-master/src/python/
2019-11-14 16:41:23.321434: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.1 instructions, but these are available on your machine and could speed up CPU computations.
2019-11-14 16:41:23.321474: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.2 instructions, but these are available on your machine and could speed up CPU computations.
2019-11-14 16:41:23.321484: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX instructions, but these are available on your machine and could speed up CPU computations.
2019-11-14 16:41:23.321501: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX2 instructions, but these are available on your machine and could speed up CPU computations.
2019-11-14 16:41:23.321514: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use FMA instructions, but these are available on your machine and could speed up CPU computations.
Initialated Mask RCNN network...
Creating net instance...
Loading net parameters...
Bus error (core dumped)
I really don't know how to solve it, could you help me?
Sorry to bothering you again

jzpwhu1996, 2019-11-15:
I have already solved the problem, thanks god.

closed 2019-11-15 by jzpwhu1996

timelessxpcn, 2019-11-29:
I meet the same problem,and my environment are python3.6.9+opencv2.4.11.when I add {PATH_TO_MASK} and {PATH_TO_OUPUT},Segmentation fault was occured,could you tell me how to solve this problem, is this problem occured due to the incorrect environment. Thank you.

hperik1, 2020-03-03:
Can this topic be opened again? I have it running when no Mask and Data paths are given, and using a RGBD TUM data set (as per example). Adding the Mask and Data paths results in a FATAL error on libprotobuf (requires 3.3.0, installed 2.6.1).  The src/python/Check is passing.
Any suggestions ?

lesurp, 2020-03-18:
@jzpwhu1996 Some explanations on what was wrong would be very much welcome, as I'm facing the same problem...

He-Rong, 2020-05-10:
> I have already solved the problem, thanks god.
I meet the same problem,and my environment are python3.6.9+opencv2.7.
I can run Check.py sucessfully.The output is :Mask R-CNN is correctly working.
But when I run the TUM rgbd example,the output is:"Bus error (core dumped)"
could you tell me how to solve this problem, is this problem occured due to the incorrect environment. Thank you.

Rushm1an, 2021-03-17:
@jzpwhu1996 how did you solve the problem? Was it just an opencv version issue?

Master-Fischer, 2022-03-01:
I had the same problem, but I was able to make DynaSLAM work correctly using nvidia-docker. Here is the link to my github repo: [https://github.com/Master-Fischer/DynaSLAM_Docker](url).
I hope it helps.

ljh192, 2022-09-25:
> Sorry to bothering you. When I run the DynaSLAM with the command "./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml Data/rgbd_dataset_freiburg3_walking_xyz/ [...]

could you please tell me how to solve this problem