BertaBescos/DynaSLAM #41: problem in running the code

wyywyywyywyy, 2020-01-14:
Hello! I have some problem in running the code. Could you please give me some suggustion?
1.When I run the DynaSLAM with the command " ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUMX3.yaml data/rgbd_dataset_freiburg3_walking_xyz/ data/rgbd_dataset_freiburg3_walking_xyz/associations.txt data/mask data/output”
My terminal's output is :
nImages: 827
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
./Examples/RGB-D/MaskSettings.yaml
RuntimeError: _ARRAY_API is not PyCapsule object
Segmentation fault (core dumped)

2.When I don't add the PATH_TO_MASKS and PATH_TO_OUTPUT
, My terminal's output is :
nImages: 3983
DynaSLAM Copyright (C) 2018 Berta Bescos, University of Zaragoza.
Input sensor was set to: RGB-D
Failed to open settings file at: Examples/RGB-D/TUMX2.yaml

wyywyywyywyy, 2020-01-14:
python version is: 3.6, opencv version is: 3.4.7.I already have run it from the DynaSLAM folder as `python src/python/Check.py.`
My terminal's output is :
Initialated Mask RCNN network...
PruebaPython
Mask R-CNN is correctly working

jingyic, 2020-02-26:
I have the same problem. Have you solved it?