BertaBescos/DynaSLAM #45: DynaSLAM build error

ndinarvand, 2020-03-11:
Hello everyone,
I compiled everything just like mentioned steps in readme.md. but I've got build error according below:

[ 62%] Building CXX object CMakeFiles/DynaSLAM.dir/src/MapDrawer.cc.o
[ 62%] Building CXX object CMakeFiles/DynaSLAM.dir/src/KeyFrameDatabase.cc.o
[ 65%] Building CXX object CMakeFiles/DynaSLAM.dir/src/PnPsolver.cc.o
[ 68%] Building CXX object CMakeFiles/DynaSLAM.dir/src/Geometry.cc.o
In file included from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1777:0,
                 from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /home/navid/DynaSLAM/include/Conversion.h:17,
                 from /home/navid/DynaSLAM/src/Conversion.cc:9:
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
In file included from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1777:0,
                 from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /home/navid/DynaSLAM/include/Conversion.h:17,
                 from /home/navid/DynaSLAM/include/MaskNet.h:26,
                 from /home/navid/DynaSLAM/src/MaskNet.cc:9:
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
/home/navid/DynaSLAM/src/Conversion.cc: In member function ‘cv::UMatData* DynaSLAM::NumpyAllocator::allocate(int, const int*, int, void*, size_t*, int, cv::UMatUsageFlags) const’:
/home/navid/DynaSLAM/src/Conversion.cc:158:25: error: ‘CV_MAX_DIM’ was not declared in this scope
         npy_intp _sizes[CV_MAX_DIM+1];
                         ^
/home/navid/DynaSLAM/src/Conversion.cc:161:13: error: ‘_sizes’ was not declared in this scope
             _sizes[i] = sizes[i];
             ^
/home/navid/DynaSLAM/src/Conversion.cc:166:13: error: ‘_sizes’ was not declared in this scope
             _sizes[dims++] = cn;
             ^
In file included from /home/navid/DynaSLAM/include/Conversion.h:17:0,
                 from /home/navid/DynaSLAM/src/Conversion.cc:9:
/home/navid/DynaSLAM/src/Conversion.cc:168:47: error: ‘_sizes’ was not declared in this scope
         PyObject* o = PyArray_SimpleNew(dims, _sizes, typenum);
                                               ^
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:135:40: note: in definition of macro ‘PyArray_SimpleNew’
         PyArray_New(&PyArray_Type, nd, dims, typenum, NULL, NULL, 0, 0, NULL)
                                        ^
In file included from /usr/local/include/opencv4/opencv2/core.hpp:54:0,
                 from /usr/local/include/opencv4/opencv2/imgproc.hpp:46,
                 from /usr/local/include/opencv4/opencv2/imgproc/imgproc.hpp:48,
                 from /home/navid/DynaSLAM/include/Conversion.h:14,
                 from /home/navid/DynaSLAM/src/Conversion.cc:9:
/home/navid/DynaSLAM/src/Conversion.cc:172:23: error: ‘CV_StsError’ was not declared in this scope
             CV_Error_(CV_StsError, ("The numpy array of typenum=%d, ndims=%d can not be created", typenum, dims));
                       ^
/home/navid/DynaSLAM/src/Conversion.cc: At global scope:
/home/navid/DynaSLAM/src/Conversion.cc:205:16: error: cannot declare variable ‘DynaSLAM::g_numpyAllocator’ to be of abstract type ‘DynaSLAM::NumpyAllocator’
 NumpyAllocator g_numpyAllocator;
                ^
/home/navid/DynaSLAM/src/Conversion.cc:74:7: note:   because the following virtual functions are pure within ‘DynaSLAM::NumpyAllocator’:
 class NumpyAllocator : public MatAllocator
       ^
In file included from /usr/local/include/opencv4/opencv2/core.hpp:59:0,
                 from /usr/local/include/opencv4/opencv2/imgproc.hpp:46,
                 from /usr/local/include/opencv4/opencv2/imgproc/imgproc.hpp:48,
                 from /home/navid/DynaSLAM/include/Conversion.h:14,
                 from /home/navid/DynaSLAM/src/Conversion.cc:9:
/usr/local/include/opencv4/opencv2/core/mat.hpp:475:23: note: 	virtual cv::UMatData* cv::MatAllocator::allocate(int, const int*, int, void*, size_t*, cv::AccessFlag, cv::UMatUsageFlags) const
     virtual UMatData* allocate(int dims, const int* sizes, int type,
                       ^
/usr/local/include/opencv4/opencv2/core/mat.hpp:477:18: note: 	virtual bool cv::MatAllocator::allocate(cv::UMatData*, cv::AccessFlag, cv::UMatUsageFlags) const
     virtual bool allocate(UMatData* data, AccessFlag accessflags, UMatUsageFlags usageFlags) const = 0;
                  ^
/home/navid/DynaSLAM/src/Conversion.cc: In member function ‘cv::Mat DynaSLAM::NDArrayConverter::toMat(const PyObject*)’:
/home/navid/DynaSLAM/src/Conversion.cc:244:17: error: ‘CV_MAX_DIM’ was not declared in this scope
     if(ndims >= CV_MAX_DIM)
                 ^
[... 177 lines cut ...]
 ^
In file included from /home/navid/DynaSLAM/src/Frame.cc:10:0:
/home/navid/DynaSLAM/include/Frame.h:128:10: warning: ‘ORB_SLAM2::Frame::mIsKeyFrame’ will be initialized after [-Wreorder]
     bool mIsKeyFrame;
          ^
/home/navid/DynaSLAM/include/Frame.h:127:13: warning:   ‘cv::Mat ORB_SLAM2::Frame::mImDepth’ [-Wreorder]
     cv::Mat mImDepth;
             ^
/home/navid/DynaSLAM/src/Frame.cc:236:1: warning:   when initialized here [-Wreorder]
 Frame::Frame(const cv::Mat &imGray, const cv::Mat &imDepth, const cv::Mat &imMask, const cv::Mat &imRGB,
 ^
/home/navid/DynaSLAM/src/Geometry.cc: In member function ‘std::vector<DynaSLAM::Geometry::DynKeyPoint> DynaSLAM::Geometry::ExtractDynPoints(const std::vector<ORB_SLAM2::Frame>&, const ORB_SLAM2::Frame&)’:
/home/navid/DynaSLAM/src/Geometry.cc:337:21: warning: unused variable ‘xIndex’ [-Wunused-variable]
                 int xIndex = minIdx.x;
                     ^
/home/navid/DynaSLAM/src/Geometry.cc: In member function ‘cv::Mat DynaSLAM::Geometry::RegionGrowing(const cv::Mat&, int&, int&, const float&)’:
/home/navid/DynaSLAM/src/Geometry.cc:1045:45: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     while(pixdist < reg_maxdist && reg_size < im.total())
                                             ^
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/DynaSLAM.dir/all' failed
make[1]: *** [CMakeFiles/DynaSLAM.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Please help me to solve....

closed 2020-03-13 by ndinarvand

cherrybonch, 2020-05-07:
Have you solved this issue? Could you please share the solution?

FrancescoFallica, 2020-07-02:
I think it's related to the OpenCV version.
Check if the CMakeLists.txt is able to find your version.