BertaBescos/DynaSLAM #48: Assertion `this->net != NULL' (again..!)

lesurp, 2020-03-19:
After encountering many issues running DynaSLAM on my machine (tensorlfow/cuda/protobuf versions...), I tried running it with Docker

Here's the backtrace when I compile in `Debug`

```
nImages: 1021
Loading Mask R-CNN. This could take a while...
Importing Mask R-CNN Settings...
    py_path: ./src/python/
    module_name: MaskRCNN
    class_name: Mask
    get_dyn_seg: GetDynSeg
[New Thread 0x7fffd288f700 (LWP 1037)]
[New Thread 0x7fffd208e700 (LWP 1038)]
[New Thread 0x7fffcd88d700 (LWP 1039)]
[New Thread 0x7fffcd08c700 (LWP 1040)]
[New Thread 0x7fffc888b700 (LWP 1041)]
[New Thread 0x7fffc808a700 (LWP 1042)]
[New Thread 0x7fffc5889700 (LWP 1043)]
[New Thread 0x7fffbcf42700 (LWP 1044)]
[New Thread 0x7fffbc741700 (LWP 1045)]
[New Thread 0x7fffb9f40700 (LWP 1046)]
[New Thread 0x7fffb573f700 (LWP 1047)]
[New Thread 0x7fffb2f3e700 (LWP 1048)]
[New Thread 0x7fffb073d700 (LWP 1049)]
[New Thread 0x7fffaff3c700 (LWP 1050)]
[Thread 0x7fffaff3c700 (LWP 1050) exited]
[Thread 0x7fffb073d700 (LWP 1049) exited]
[Thread 0x7fffb2f3e700 (LWP 1048) exited]
[Thread 0x7fffb573f700 (LWP 1047) exited]
[Thread 0x7fffb9f40700 (LWP 1046) exited]
[Thread 0x7fffbc741700 (LWP 1045) exited]
[Thread 0x7fffbcf42700 (LWP 1044) exited]
[Thread 0x7fffc5889700 (LWP 1043) exited]
[Thread 0x7fffc808a700 (LWP 1042) exited]
[Thread 0x7fffc888b700 (LWP 1041) exited]
[Thread 0x7fffcd08c700 (LWP 1040) exited]
[Thread 0x7fffcd88d700 (LWP 1039) exited]
[Thread 0x7fffd208e700 (LWP 1038) exited]
[Thread 0x7fffd288f700 (LWP 1037) exited]
/usr/local/lib/python2.7/dist-packages/h5py/__init__.py:40: UserWarning: h5py is running against HDF5 1.10.0 when it was built against 1.10.4, this may cause problems
  '{0}.{1}.{2}'.format(*version.hdf5_built_version_tuple)
Using TensorFlow backend.
Initializing Mask RCNN network...
./src/python
[New Thread 0x7fffc5889700 (LWP 1054)]
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/keras/backend/tensorflow_backend.py:4070: The name tf.nn.max_pool is deprecated. Please use tf.nn.max_pool2d instead.

WARNING:tensorflow:From /DynaSLAM/src/python/model.py:366: where (from tensorflow.python.ops.array_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.where in 2.0, which has the same broadcast rule as np.where
WARNING:tensorflow:From /DynaSLAM/src/python/model.py:390: calling crop_and_resize_v1 (from tensorflow.python.ops.image_ops_impl) with box_ind is deprecated and will be removed in a future version.
Instructions for updating:
box_ind is deprecated, use box_indices instead
WARNING:tensorflow:From /DynaSLAM/src/python/model.py:746: py_func (from tensorflow.python.ops.script_ops) is deprecated and will be removed in a future version.
Instructions for updating:
tf.py_func is deprecated in TF V2. Instead, there are two
    options available in V2.
    - tf.py_function takes a python function which manipulates tf eager
    tensors instead of numpy arrays. It's easy to convert a tf eager tensor to
    an ndarray (just call tensor.numpy()) but having access to eager tensors
    means `tf.py_function`s can use accelerators such as GPUs as well as
    being differentiable using a gradient tape.
    - tf.numpy_function maintains the semantics of the deprecated tf.py_func
    (it is not differentiable, and manipulates numpy arrays). It drops the
    stateful argument making all functions stateful.
    
rgbd_tum: /DynaSLAM/src/MaskNet.cc:43: DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->net != NULL' failed.

Thread 1 "rgbd_tum" received signal SIGABRT, Aborted.
[... 109 lines cut ...]
2020-03-19 18:06:27.435172: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1005] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-03-19 18:06:27.435636: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1763] Adding visible gpu devices: 0
2020-03-19 18:06:27.435707: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0
2020-03-19 18:06:27.436585: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1181] Device interconnect StreamExecutor with strength 1 edge matrix:
2020-03-19 18:06:27.436600: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1187]      0 
2020-03-19 18:06:27.436610: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1200] 0:   N 
2020-03-19 18:06:27.436711: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1005] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-03-19 18:06:27.437166: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1005] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-03-19 18:06:27.437596: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1326] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 2965 MB memory) -> physical GPU (device: 0, name: GeForce GTX 1050 Ti, pci bus id: 0000:01:00.0, compute capability: 6.1)
Initialated Mask RCNN network...
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/keras/backend/tensorflow_backend.py:422: The name tf.global_variables is deprecated. Please use tf.compat.v1.global_variables instead.

2020-03-19 18:06:31.597647: W tensorflow/compiler/jit/mark_for_compilation_pass.cc:1412] (One-time warning): Not using XLA:CPU for cluster because envvar TF_XLA_FLAGS=--tf_xla_cpu_global_jit was not set.  If you want XLA:CPU, either set that envvar, or use experimental_jit_scope to enable XLA:CPU.  To confirm that XLA is active, pass --vmodule=xla_compilation_cache=1 (as a proper command-line flag, not via TF_XLA_FLAGS) or set the envvar XLA_FLAGS=--xla_hlo_profile.
2020-03-19 18:06:31.668133: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcublas.so.10.0
2020-03-19 18:06:31.816810: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudnn.so.7
Mask R-CNN is correctly working
```

You can find the `Dockerfile` as well as the API updates on my [fork](https://github.com/lesurp/DynaSLAM/tree/tf_keras_updates)

I saw issues #24, but my `Check.py` works fine
#11 as well, but I run everything from the root directory:
`./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml /data/tum/rgbd_dataset_freiburg3_walking_halfsphere /data/tum/rgbd_dataset_freiburg3_walking_halfsphere/associations.txt no_save` 

I'm seeing comments about opencv versions (my `Dockerfile` uses opencv3), I will update this issue whenever I find a solution...

PushyamiKaveti, 2020-07-09:
did you get DynaSLAM to run finally?

liubamboo, 2021-04-12:
Firstly I run DynaSLAM ok in the teminal, and then I run in the ide(CLion), it shows the bug:
mono_tum: /home/n609/project/DynaSLAM/src/MaskNet.cc:39: DynaSLAM::SegmentDynObject::SegmentDynObject(): Assertion `this->py_module != NULL' failed.

Finally I change the working directory to DynaSLAM and now it can work.
[image]