BertaBescos/DynaSLAM #51: Software requirement....

ndinarvand, 2020-04-28:
Hello,
Would you please help me to install and run Dynaslam on ubuntu by introducing software version:
Ubuntu version...
Cuda cersion...
Tensorflow Version...
Keras version...

Thanks a lot.

staghavi93, 2020-04-30:
Hello, 
I also, have the same problem and I don't know the source code needs for instance Tensorflow or Tensorflow-gpu. I would appreciate any help.

dinarkino, 2020-06-16:
Hi,
Thank you for the repository! 
I also agree about the requirements. It's not clear which version of Keras and TensorFlow to use. After installation of Keras with pip I got [this](https://github.com/matterport/Mask_RCNN/issues/694) error, then after reinstalling it with
`pip install 'keras==2.1.6' --force-reinstall` 
I finally got "Mask R-CNN is correctly working". Now it's time to solve Bus error (core dumped) :))

Here is another question, I found in several issues here (more than one year old) that it's better to use opencv 2.4.11 is that still true? For now, I tried everything with 3.4.0.

It seems that there is a need in a more detailed Getting Started guide. 
@BertaBescos

dinarkino, 2020-06-16:
Another interesting thing that actually I was able to build DynaSLAM without Keras and TensorFlow, which I occasionally installed for the python 3. Also, there is such an error in building process:

```
Cannot find source file:
    Examples/Monocular/mono_carla.cc
...
CMake Error: CMake can not determine linker language for target: mono_carla
CMake Error: Cannot determine link language for target "mono_carla".
...
make[2]: *** No rule to make target 'CMakeFiles/mono_carla.dir/build'.  Stop.
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/mono_carla.dir/all' failed
```

I'm not sure is that right behaviour or not because finally building is successful. And when I clone repository this file also exist in Examples/Monocular/.

KaninchenM, 2020-06-17:
I've installed successfully (reference: `https://blog.csdn.net/Mac0327/article/details/105861895`). Now share my version with you.
Ubuntu 16.04
CUDA 9.0
python 2.7
tensorflow-gpu 1.12.3
keras 2.0.9
opencv 2.14.11 (which should modify some code, reference here: `https://pan.baidu.com/share/init?surl=9AaTZUMqCanJFHU2tAGetw`, pw:`ph4w` ) 
Hope this helps.

ndinarvand, 2020-06-22:
Hi
I followed instructions in https://blog.csdn.net/Mac0327/article/details/105861895
But in tensorflow installation it needs python>=3.6

Please help to solve it.

dinarkino, 2020-06-23:
@KaninchenM is there a way to download changed version of opencv 2.14.11 without Baidu? I can't understand how to do that without an account

KaninchenM, 2020-06-29:
I use this url : https://pan.baidu.com/share/init?surl=9AaTZUMqCanJFHU2tAGetw
and the password is ph4w
The author is here: https://blog.csdn.net/ljl1015ljl/article/details/100749835

dinarkino, 2020-06-29:
> I use this url : https://pan.baidu.com/share/init?surl=9AaTZUMqCanJFHU2tAGetw and the password is ph4w The author is here: https://blog.csdn.net/ljl1015ljl/article/details/100749835
> […](#)

Yeah, I mean even using this link it asks for downloading Baidu app and to log in there using an account which I can't create. If it's possible to download this archive to some other cloud, like dropbox, it will be very helpful. 

By the way, I tried building OpenCV 2.4.11 several times using many advices and I still can't do it without errors. Seems like this version is already unsupported and it will be much better if DynaSLAM will work with OpenCV 3.

KaninchenM, 2020-06-29:
> I use this url : https://pan.baidu.com/share/init?surl=9AaTZUMqCanJFHU2tAGetw and the password is ph4w The author is here: https://blog.csdn.net/ljl1015ljl/article/details/100749835
>  …

From the issues in github project BertaBescos/DynaSLAM, I think it have not support opencv3.x well, and have not support python3.x well either. 
There may be some restrictions on Baidu network disk in your web environment. I send the tar file I've downloaded from the Baidu network disk. Hope it can help you.

 
 

从QQ邮箱发来的超大附件

opencv-2.4.11.tar.gz (86.66M, 2020年07月29日 14:39 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=2a6531374346f79f34e0a3234738504a5751020e565d51514e525455571556005a071c0e525e014854525452030b060300570802612c620a13005f541715504b574b00064f4c03174d024b375c&code=ce17a8be

dinarkino, 2020-07-02:
@KaninchenM thank you so much for the link! I finally have built opencv 2.4.11 without any errors. Now I have another error but it's another story :)

KaninchenM, 2020-07-17:
> Hi
> I followed instructions in https://blog.csdn.net/Mac0327/article/details/105861895
> But in tensorflow installation it needs python>=3.6
> 
> Please help to solve it.

Have you solved the problem? I installed env using anaconda, so my command are:
`conda install tensorflow-gpu=1.12.3`
I've tried to install it without anaconda. The following works. But downloading the package online may very slow even throw network error. It's better for you to download them from reliable source before installing.
```
pip install numpy-1.13.3-cp27-cp27mu-manylinux1_x86_64.whl
pip install setuptools-44.1.1-py2.py3-none-any.whl -i https://pypi.tuna.tsinghua.edu.cn/simple/
pip install Markdown-3.2.1-py2.py3-none-any.whl -i https://pypi.tuna.tsinghua.edu.cn/simple/
pip install tensorflow_gpu-1.12.3-cp27-cp27mu-manylinux1_x86_64.whl -i https://pypi.tuna.tsinghua.edu.cn/simple/
```

shubhendu1907, 2020-07-30:
> Have you solved the problem? I installed env using anaconda, so my command are:
> `conda install tensorflow-gpu=1.12.3`
> I've tried to install it without anaconda. The following works. But downloading [...]

Worked with this configuration:
Ubuntu 16.04
CUDA 10.0
python 2.7
tensorflow 1.12.3
keras 2.0.9
opencv 2.14.11

But get Segmentation Fault error with tensorflow-gpu 1.12.3.
Do you know what might be causing this?

KaninchenM, 2020-08-03:
> Worked with this configuration:
> Ubuntu 16.04
> CUDA 10.0
> python 2.7
> tensorflow 1.12.3
> keras 2.0.9
> opencv 2.14.11
>
> But get Segmentation Fault error with tensorflow-gpu 1.12.3.
> Do you know what might be [...]

Where the error appears? I mean which line of code. Solving the problem may need more information.

maclo4, 2020-12-20:
> Worked with this configuration:
> Ubuntu 16.04
> CUDA 10.0
> python 2.7
> tensorflow 1.12.3
> keras 2.0.9
> opencv 2.14.11
>
> But get Segmentation Fault error with tensorflow-gpu 1.12.3.
> Do you know what might be [...]

I had this same issue.

KaninchenM, 2020-12-21:
> I had this same issue.

Many problems may cause` Segmentation Fault error`. It's better to locate your error to certain line of code.

maclo4, 2020-12-21:
> Many problems may cause` Segmentation Fault error`. It's better to locate your error to certain line of code.

[image]

This is all the error shows

maclo4, 2020-12-21:
> Many problems may cause` Segmentation Fault error`. It's better to locate your error to certain line of code.
[image]

This is all that the error shows

KaninchenM, 2020-12-22:
> This is all that the error shows

It looks like you haven't install maskrcnn module successfully. Maybe you can refer to this:https://blog.csdn.net/Mac0327/article/details/105861895
especially see the `2.2` and `2.2.5` part to check.

maclo4, 2020-12-22:
> It looks like you haven't install maskrcnn module successfully. Maybe you can refer to this:https://blog.csdn.net/Mac0327/article/details/105861895
> especially see the `2.2` and `2.2.5` part to check.

Mask_RCNN works when I use normal tensorflow instead of tensorflow-gpu. I actually followed that same guide for setting up DynaSLAM but ended up getting really slow results when using the masks option. So that's why I was trying tensorflow-gpu just hoping that it'd speed up the process. But now I've realized that my nvidia drivers arent set up properly to allow WSL2 to use CUDA so I'm wondering if maybe that is why it's running so slow. Any ideas for speeding up the image processing on WSL2?

Skywalker666666, 2021-04-07:
> Examples/Monocular/mono_carla.cc

How did you find out the file?