BertaBescos/DynaSLAM #52: OpenCV Error: Assertion failed (a_size.width == len) in gemm

KaninchenM, 2020-06-17:
I've installed the environment of DynaSlam on two servers. Because not only me use the servers, I use docker and anaconda separately on the two servers to prepare the environment. I've check the maskrcnn's environment and compiled the DynaSlam code successfully both on the two. However, when I ran the rgbd_tum by same command, the one on docker successfully completed, while the one using anaconda(install the python packages by conda, install the c++ dependence by apt-get or compile the source code) got error:
`OpenCV Error: Assertion failed (a_size.width == len) in gemm, file /home/tensorflow03/mxw_slam/opencv-2.4.11/modules/core/src/matmul.cpp, line 728
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/tensorflow03/mxw_slam/opencv-2.4.11/modules/core/src/matmul.cpp:728: error: (-215) a_size.width == len in function gemm

Aborted (core dumped)
`
The data is Tum-RGBD, and I've tried the `rgbd_dataset_freiburg3_walking_static` and `rgbd_dataset_freiburg3_walking_xyz`. Let the program print the count of passing image to locate the error.  In sequence `rgbd_dataset_freiburg3_walking_xyz`, error appeared at 775th/828(sum of images), while in `rgbd_dataset_freiburg3_walking_static`  appeared at 599th/718.
I've set the opencv directory in `CMakeLists.txt`, and the error seems show it was using opencv-2.4.11. Why the same code shows different results? How should I fix it?

KaninchenM, 2020-06-17:
I've check the CMakeLists.txt in folder `/DynaSLAM` , `/DynaSLAM/Thirdparty/DBoW2/` and `/DynaSLAM/Thirdparty/g2o/`, and modifed `SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native") ` to
`SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3") `
 then compiled the code. It fix my problem.

KaninchenM, 2020-06-17:
Remember, if someone get this error when using only multi-view, or get `Segmentation fault(core error)` using maskrcnn, please make sure you're running code under `/DynaSLAM` folder. Beacause when I try to run it outside the folder, I meet the error. It may be a point relative with C++, I guess(it's my first time use C++)

dinarkino, 2020-07-02:
@KaninchenM I got this error which is very similar to yours: 

```
OpenCV Error: Assertion failed (size.width>0 && size.height>0) in imshow, file /home/telekprod/packages/opencv-2.4.11/modules/highgui/src/window.cpp, line 261
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/telekprod/packages/opencv-2.4.11/modules/highgui/src/window.cpp:261: error: (-215) size.width>0 && size.height>0 in function imshow

Aborted (core dumped)
```

I tried to recompile everything without -march=native but nothing changed (I'm not sure should I change that only for CMAKE_CXX_FLAGS_RELEASE or also for CMAKE_CXX_FLAGS and CMAKE_C_FLAGS). I'm running the code for the mono_tum from DynaSLAM folder. I found on the Stack that this error often happens with the wrong path to the images, or for the empty files, but it seems that I'm using the same code as in readme and there are actually images at the path

```
./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM2.yaml ~/data/t_desk

ls ~/data/t_desk
accelerometer.txt  depth  depth.txt  groundtruth.txt  rgb  rgb.txt
```

KaninchenM, 2020-07-17:
> @KaninchenM I got this error which is very similar to yours:
>
> ```
> OpenCV Error: Assertion failed (size.width>0 && size.height>0) in imshow, file [...]

Sorry to see it now. Have you solve the problem? I didn't run the mono scripts, I ran the `rgbd_tum.cc`. Maybe you can check the following 3 points:
1. Change not only `CMAKE_CXX_FLAGS_RELEASE ` but also like 
```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3")
SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3") 
SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3") 
```
2. Change not only the `/DynaSLAM/CMakeLists.txt` but also `/DynaSLAM/Thirdparty/DBoW2/CMakeLists.txt` and `/DynaSLAM/Thirdparty/g2o/CMakeLists.txt`
3. Use the absolute path in your command.

liubamboo, 2021-04-07:
I have the same bug. But the method above can't solve my bug. And the bug sometimes appears and sometimes not. it's very strange

KaninchenM, 2021-04-12:
> I have the same bug. But the method above can't solve my bug. And the bug sometimes appears and sometimes not. it's very strange

What's the difference between the states(the bug appears or not)? Maybe the different shape of input img? If there's no change, maybe the problem caused in initialization.

liubamboo, 2021-04-13:
when I run rgbd_tum on fr3_s_xyz sequence. it appear a bug:
Light Tracking not working because Tracking is not initialized...
Geometry not working.
New map created with 268 points
Geometry not working.
Geometry not working.
Geometry not working.
Geometry not working.
Geometry not working.
Geometry not working.
Geometry not working.
OpenCV Error: Assertion failed (a_size.width == len) in gemm, file /home/n609/install/opencv-2.4.11/modules/core/src/matmul.cpp, line 728
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/n609/install/opencv-2.4.11/modules/core/src/matmul.cpp:728: error: (-215) a_size.width == len in function gemm

Hello-Water, 2021-08-16:
> @KaninchenM I got this error which is very similar to yours:
>
> ```
> OpenCV Error: Assertion failed (size.width>0 && size.height>0) in imshow, file [...]

This error often occurs at the end of reading a video, because after a video is read and displayed, the image is empty. Maybe you can make the following changes in src/Viewer.cc:
`if(!im.empty()) cv::imshow("DynaSLAM: Current Frame",im);`
`if(!im_dyn.empty()) cv::imshow("DynaSLAM: Dynamic Frame", im_dyn);`