BertaBescos/DynaSLAM #54: How long it takes to run a sequence such as Tum RGBD fr3_walking_rpy?

KaninchenM, 2020-07-03:
In paper, it shows Timing Analysis in Chapter IV. The table shows the average computational time of Multi-view Geometry and Background Inpainting. Meanwhile, quote the maskrcnn time from other's paper. It shows in sequence Tum RGBD fr3_walking_rpy, Multi-view Geometry needs 235.98ms/frame, Background Inpainting needs 183.56ms/frame, Maskrcnn needs 195ms/frame. Those needs at least 615ms/frame(though the result had not give the parameters such as `ORBextractor.nFeatures`). In my experiment(GPU is Tesla P4, `ORBextractor.nFeatures:1000`), Multi-view Geometry needs 170-210ms/frame, Background Inpainting needs 300-320ms/frame, Maskrcnn needs 320ms/frame. Those needs at least 850ms/frame. 
Could you please share your time analysis result more? For example, how many seconds did you spend to run the whole program? Could you please give some suggestions to speed up the model to make the program real time? 
Thank you.

KaninchenM, 2020-07-16:
Using the code (branch carla) to run 4 sequences: TUM RGBD's fr3_walking_xyz, fr3_walking_rpy, fr3_walking_static, fr3_walking_halfsphere. Comparing the time analysis result in paper, in my experiment, the  `InpaintFrame` part took 2 times longer than the paper. Does anyone run the time as the paper show?
[image]
Use Blue to mark the result of paper, Yellow to mark my result.

maclo4, 2020-12-20:
When using masks, the program takes 20 minutes to finish at least for me

KaninchenM, 2020-12-21:
> When using masks, the program takes 20 minutes to finish at least for me

Which dataset or sequence did you use?

maclo4, 2020-12-21:
> > When using masks, the program takes 20 minutes to finish at least for me
> 
> Which dataset or sequence did you use?

I downloaded "rgbd_dataset_freiburg1_xyz"

Rushm1an, 2021-03-19:
> When using masks, the program takes 20 minutes to finish at least for me

Hi, could you please tell me about your system configurations? And in general, how much GPU is required for performing the experiment on TUM datasets?

liubamboo, 2021-04-07:
@KaninchenM I use i7 cpu to run the multi-view geometry method on TUM RGBD walking_xyz. and I get the tracking time:
median tracking time: 0.0728491
mean tracking time: 2.43465
it is very slow. 
are you same as these?

liubamboo, 2021-04-07:
@KaninchenM How can you get the time of  each part? thanks a lot

KaninchenM, 2021-04-12:
> @KaninchenM I use i7 cpu to run the multi-view geometry method on TUM RGBD walking_xyz. and I get the tracking time:
> median tracking time: 0.0728491
> mean tracking time: 2.43465
> it is very slow.
> are you same as these?

Almost same. Even when I used GPU to speed up the process of Maskrcnn part, it still can not be real-time.

KaninchenM, 2021-04-12:
> @KaninchenM How can you get the time of each part? thanks a lot

I don't know if my method is correct, but I added some variable in certain code block to calculate the running time.