BertaBescos/DynaSLAM #61: What's the meaning of parallax angle α(the angle between the back-projections of x and x`)

KaninchenM, 2020-12-18:
What's the meaning of parallax angle α (the angle between the back-projections of x and x`) ? I'm not sure what the `back-projections` mean? Does it mean a method of finding features by Histogram model? But the source code do not mention anything about Histogram model.
Is the α just the angle between X-x and X-x` like Fig,3? If so, when I change data set, how can I evaluate which parallax angles threshold is better?
[image]
[image]
Waiting for your reply. Thank you!

jmfacil, 2020-12-18:
Hi, 

As you say the angle alpha is the angle between between the rays X-KF and X-CF. That threshold on alpha should not vary much between datasets, you can leave it in 30. Since we are using depth variation to distinguish dynamic elements, we need to filter those variations that are due to normal occlusion of static elements. 

The way we do it is by putting a threshold in the parallax angle. If it is greater than 30 then occlusion may be due to static elements and not  dynamic. So we cant consider them dynamic. 

Best
Chema

KaninchenM, 2020-12-21:
> Hi,
>
> As you say the angle alpha is the angle between between the rays X-KF and X-CF. That threshold on alpha should not vary much between datasets, you can leave it in 30. Since we are using depth [...]

Thank you for your reply. But I'm still not sure what's the `back-projections of x` mean? From the words in paper, I can't understand so I can't draw the α in picture when the KP is occluded by a normal static object :(

KaninchenM, 2020-12-21:
> Thank you for your reply. But I'm still not sure what's the `back-projections of x` mean? From the words in paper, I can't understand so I can't draw the α in picture when the KP is occluded by a [...]

I'm still thinking about it. In fig(b), how we find the x'(blue +)? 
A. By matching the descriptions of X, find x' in feature points in CF image.
Which one is right?
B. By transform pose from KF's x depending on Light Track's result.

KaninchenM, 2020-12-21:
Another question I'v mentioned in [issue-56](https://github.com/BertaBescos/DynaSLAM/issues/56) Q4 is:
τ = 0.4m in paper but
τ = 0.6m in code .
Why and how to deal with it? How to set the τ in other dataset?

jmfacil, 2020-12-21:
> > > Hi,
> > > As you say the angle alpha is the angle between between the rays X-KF and X-CF. That threshold on alpha should not vary much between datasets, you can leave it in 30. Since we are using depth variation to distinguish dynamic elements, we need to filter those variations that are due to normal occlusion of static elements.
> > > The way we do it is by putting a threshold in the parallax angle. If it is greater than 30 then occlusion may be due to static elements and not dynamic. So we cant consider them dynamic.
> > > Best
> > > Chema
> > 
> > 
> > Thank you for your reply. But I'm still not sure what's the `back-projections of x` mean? From the words in paper, I can't understand so I can't draw the α in picture when the KP is occluded by a normal static object :(
> 
> I'm still thinking about it. In fig(b), how we find the x'(blue +)?
> A. By matching the descriptions of X, find x' in feature points in CF image.
> B. By transform pose from KF's x depending on Light Track's result.
> Which one is right?

B, we know the pose from the light tracking.

jmfacil, 2020-12-21:
> Thank you for your reply. But I'm still not sure what's the `back-projections of x` mean? From the words in paper, I can't understand so I can't draw the α in picture when the KP is occluded by a [...]

Back-projection it refers to the point in the 3D from its projected view in the image and depth

KaninchenM, 2020-12-21:
> B, we know the pose from the light tracking.

Thank you so much. Happy to refresh this web page and see your reply.
In my opinion, there're two types of `Occlusion` which may let the X occluded: one is dynamic, another one normal static. The α is used to distinguish the two types. When α>=30°, the X is identified as `occluded by normal static object`.
But I still can understand the principle of this strategy.

jmfacil, 2020-12-21:
If the parallax angle is less than 30 (α>=30°), the camera has not moved as much, and therefor if there is an occlusion it is likely to be produced by a dynamic element in the scene. When it is bigger than 30 the occlusion could be just be a normal occlusion.

KaninchenM, 2020-12-21:
> If the parallax angle is less than 30 (α>=30°), the camera has not moved as much, and therefor if there is an occlusion it is likely to be produced by a dynamic element in the scene. When it is bigger than 30 the occlusion could be just be a normal occlusion.

Sorry, ` parallax angle is less than 30` means `α>=30°`? I think `parallax angle is less than 30` means `α<30°`

jmfacil, 2020-12-21:
Edited. Thanks

KaninchenM, 2020-12-22:
> Edited. Thanks

Thank you very much. Solve my doubts which lasted for months :)

KaninchenM, 2020-12-22:
> Another question I'v mentioned in [issue-56](https://github.com/BertaBescos/DynaSLAM/issues/56) Q4 is:
> τ = 0.4m in paper but
> τ = 0.6m in code .
> Why and how to deal with it? How to set the τ in other dataset?

Oh here is another related issue. I had understood more about the code of `Geometry.cc`. Could you give me some suggestions about the value of τ.

wrotcat, 2022-04-05:
@jmfacil  Hi, sorry to bother you. I am confused about a question in the code recently. I understood the idea about parallax angle. In code `Geometry::ExtractDynPoints()`, why the T_{cw} of current frame and keyframe are used to calculate  X-KF and X-CF instead of T_{wc}? X here is the mappoint in world coordinates. So we should use T_{wc} to express the world coordinates of KF and CF? Thanks for your help in advance!

Echo-gh, 2024-06-12:
> @jmfacil Hi, sorry to bother you. I am confused about a question in the code recently. I understood the idea about parallax angle. In code `Geometry::ExtractDynPoints()`, why the T_{cw} of current [...]

Hey, I'm bothered by the same question. Have u figured it out yet?

KaninchenM, 2024-06-12:
您的邮件已收到，我会尽快回复您！