BertaBescos/DynaSLAM #65: Segmentation fault when run PyArray_Check(o) in NDArrayConverter::toMat in src/Conversion.cc

linpeisensh, 2021-01-15:
The result is shown in the following image:
[image]
The MaskRCNN is initialized successfully, but Segmentation fault occurs, when run PyArray_Check(o) in NDArrayConverter::toMat in src/Conversion.cc.
I have checked NDArrayConverter::init() run before this PyArray_Check(o), that means I have "Py_Initialize();" and " _import_array();".
[image]

Could someone tell me how to solve this problem?

closed 2021-03-03 by linpeisensh

Skywalker666666, 2021-04-14:
Excuse me, How you solved this problem?

Skywalker666666, 2021-04-14:
@linpeisensh  how do you figure out this one?

linpeisensh, 2021-04-14:
Sorry, I have no idea how to solve it.