BertaBescos/DynaSLAM #67: How can I only use Mask R-CNN segmentation to detect dynamic objects?

liubamboo, 2021-04-07:
I want to verify the output below
[image]

it only use Mask R-CNN segmentation to detect dynamic objects, but the parameter of main function can't support it. it only support only multi-view geometry method(G), both geometrical and deep learning approaches(N+G). But It can't support only deep learning method(G). Can you help me? thank you very much.

hongu0603, 2021-07-21:
> I want to verify the output below
> [image]
>
> it only use Mask R-CNN segmentation to detect dynamic objects, but the parameter of main function can't support it. it only support only multi-view [...]

i have the same question with you,did you solve it ?

Hello-Water, 2021-08-17:
> I want to verify the output below
> [image]
>
> it only use Mask R-CNN segmentation to detect dynamic objects, but the parameter of main function can't support it. it only support only multi-view [...]

Hello, how to set 'PATH_TO_MASKS`，like this: "DynaSLAM/src/python/mask_rcnn_coco.h5" ? But it does not work, please give me some suggestions~ my environment:ubuntu18+opencv3.2+tensorflow1.14_keras2.2.5

wenyan96, 2021-12-04:
you can modify the code in Tracking::GrabImageRGBD() with reference of ORBSLAM2. but remember maintain the imMask variable!
And did you test the onlyGeometry code? when i tried this, i got a segment default error.