BertaBescos/DynaSLAM #78: Segmentation Fault when invoking  GetDynSeg

y9luiz, 2022-01-06:
Hello, I'm trying to run this project,  for the geometric approach everything is working but for mask-rcnn approach it seens that the numpy array is not been serealized correctly, when I call GetDynSeg at line 672 of the file `src/MaskNet.cc`

**PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image);**

I have a segmentation fault.

Anyone knows how solve it? I already tried to debug from python side and I  even couldn't print the type of image argument, it's very strange, my conclusion was that this numpy array is not been correctly serialized at **NDArrayConverter::toNDArray()** function

y9luiz, 2022-01-06:
the src/python/Check.py is working nicely

y9luiz, 2022-01-06:
OS info: Ubuntu 20.04 x86_64
OpenCV Version: 3.4.7
Numpy Version: 1.21.5
g++ Version: 7.5.0

y9luiz, 2022-01-07:
I solve this problem downgrading to opencv 2.4

alexs7, 2022-01-16:
@y9luiz hello, have you got DynaSLAM working now ? can I ask what we should pass for the "PATH_TO_MASKS" ?

y9luiz, 2022-01-16:
> @y9luiz hello, have you got DynaSLAM working now ? can I ask what we should pass for the "PATH_TO_MASKS" ?

Yeah @alexs7 , now it's working, the parameter PATH_TO_MASKS is the folder that u want to store the masks generated by the algorithm

alexs7, 2022-01-16:
I run this:

`./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml ../datasets/TUM/rgbd_dataset_freiburg3_walking_rpy/ Examples/RGB-D/associations/fr3_walking_rpy.txt Examples/RGB-D/masks_temp Examples/RGB-D/masks_temp_output
`

and I get this: 

> Creating net instance...
> Loading net parameters...
> [ WARN:0@5.561] imread_('no_save/no_file'): can't open/read file: check file path/integrity
> [1]    17264 bus error (core dumped)  ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM3.yaml   

Did you have this error ?

y9luiz, 2022-01-16:
I have no ideia about your error.

Are you alredy tried to execute the script 
`src/python/Check.py` ?

this script should test your python enviroment and check if is well configured

alexs7, 2022-01-18:
@y9luiz sorry for the late reply. here is what I get when I run that script:

> python2.7 src/python/Check.py
> /home/alex/.local/lib/python2.7/site-packages/pycocotools/coco.py:49: UserWarning: 
> This call to matplotlib.use() has no effect because the backend has already
> been chosen; matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
> or matplotlib.backends is imported for the first time.
> 
> The backend was *originally* set to 'TkAgg' by the following code:
>   File "src/python/Check.py", line 8, in <module>
>     import matplotlib.pyplot as plt
>   File "/home/alex/.local/lib/python2.7/site-packages/matplotlib/pyplot.py", line 71, in <module>
>     from matplotlib.backends import pylab_setup
>   File "/home/alex/.local/lib/python2.7/site-packages/matplotlib/backends/__init__.py", line 17, in <module>
>     line for line in traceback.format_stack()
> 
> 
>   import matplotlib; matplotlib.use('Agg')
> Using TensorFlow backend.
> Initializing Mask RCNN netwFork...
> ./src/python
> WARNING:tensorflow:From /home/alex/.local/lib/python2.7/site-packages/tensorflow/python/framework/op_def_library.py:263: colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
> Instructions for updating:
> Colocations handled automatically by placer.
> WARNING:tensorflow:From /home/alex/.local/lib/python2.7/site-packages/keras/backend/tensorflow_backend.py:3144: calling dropout (from tensorflow.python.ops.nn_ops) with keep_prob is deprecated and will be removed in a future version.
> Instructions for updating:
> Please use `rate` instead of `keep_prob`. Rate should be set to `rate = 1 - keep_prob`.
> WARNING:tensorflow:From /home/alex/DynaSLAM/src/python/model.py:746: py_func (from tensorflow.python.ops.script_ops) is deprecated and will be removed in a future version.
> Instructions for updating:
> tf.py_func is deprecated in TF V2. Instead, use
>     tf.py_function, which takes a python function which manipulates tf eager
>     tensors instead of numpy arrays. It's easy to convert a tf eager tensor to
>     an ndarray (just call tensor.numpy()) but having access to eager tensors
>     means `tf.py_function`s can use accelerators such as GPUs as well as
>     being differentiable using a gradient tape.
>     
> 2022-01-18 13:25:28.892448: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
> 2022-01-18 13:25:28.912843: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 3199980000 Hz
> 2022-01-18 13:25:28.913257: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x56177eebd2e0 executing computations on platform Host. Devices:
> 2022-01-18 13:25:28.913277: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): <undefined>, <undefined>
> Initialated Mask RCNN network...
> Mask R-CNN is correctly working

alexs7, 2022-01-18:
@y9luiz except some warning this shouls be OK right ? What do you get ?

y9luiz, 2022-01-19:
@alexs7  the python environment its okay, did you are using the opencv 2.4? how I said in this topic, my issues stopped after downgrade the opencv version

alexs7, 2022-01-20:
@y9luiz yes fixed it by downgrading to opencv 2.4

did you get the inpainting code working?

y9luiz, 2022-01-20:
@alexs7 no I never try to use the impainting functionality

alexs7, 2022-01-24:
@y9luiz I got my own fork here with updated details on how to build: https://github.com/alexs7/DynaSLAM

y9luiz, 2022-01-24:
Nice job @alexs7!

upperfish, 2024-03-13:
> Hello, I'm trying to run this project and for the mask-rcnn approach I'm facing some troubles ...
> 
> A unexpected segmentation fault occurs when I try to run with the Mask-RCNN enabled and after a few debugging I found this:
> 
> the problem occurs when I call GetDynSeg at line 62 of the file `src/MaskNet.cc`
> 
> `PyObject* py_mask_image = PyObject_CallMethod(this->net, const_cast<char*>(this->get_dyn_seg.c_str()),"(O)",py_image);`
> 
> this line jumps to the python script `src/python/MaskRCNN.py`, more specificly to function
> 
> `def GetDynSeg(self, image,image2=None)`
> 
> the function expect that the second argument (image) must be a numpy array like any other python OpenCV image, but
> 
> when the code try to get the content from the image it will produce a segmentation fault
> 
> My conclusion was that numpy array is not been correctly serialized at **NDArrayConverter::toNDArray()** function

do you slove? i have same problem? please help god!