BertaBescos/DynaSLAM #87: All masks pics output is pure black

shanpenghui, 2022-08-04:
Tks for wonderful work! 
But I have the problem, that when I run this work with dataset.
All the masks pics in output folder is black, and my settings seem right.
Program is also running normally and didn't crash.
I don't know its reason
[image]
[image]
.

zhangzjjjjjj, 2022-10-26:
I have the same problem, would you know the answer now?

WWWenZhan, 2025-05-22:
I met the same situation, how did you solve it?