BertaBescos/DynaSLAM #93: Arxiv review version said The project code will be released upon acceptance. [The RAL published version deleted this sentence]

Kin-Zhang, 2022-12-10:
Here are two screenshots of the paper on review and publication. You deleted quite an important sentence (which promised public code on review) on the RAL version...

[image]

[image]

I think if the next time you don't want to open-source, there is no need to write this sentence in the version under review.

Kin-Zhang, 2022-12-10:
Related issues: https://github.com/BertaBescos/DynaSLAM/issues/91, https://github.com/BertaBescos/DynaSLAM/issues/92

Hello-Water, 2023-01-05:
@Kin-Zhang 
It mentioned in RAL that DynaSLAM Ⅱ is an open-source SLAM system.
[image]

closed 2023-04-19 by Kin-Zhang